veebch/btcticker #1: Fallback URLs not functioning

rossdonald97, 2020-12-02:
Hi, great script, using it right now on the same e-paper display! 

Noticed the fallback URL: "http://llvll.ch/fallbackurlhistoric.json" returns an HTTP 404 on connection. (Line 59 in btcticker.py).

The same goes for the "http://llvll.ch/fallbackurllive.json" (Line 75 in btcticker.py). For the urllive I also noticed line 76 `rawlivecoin = requests.get(livecoinapi).json()` - should livecoinapi not be "fallbackurl" as in the previous fallback exception?

In the script the URLs are HTTP not HTTPS as well so changing that might be worth it since its a redirect anyway.

Thanks!

llvllch, 2020-12-02:
Hey Ross!

And here was me thinking that I was chucking code into the void :)

Those two urls will have content in as soon as I push the R script that
takes care of it. It's just there as a backup for the lights going off at
coinapi.

There is a very high chance that your coding skills eclipse mine (I'm a
mathematician). All (and I mean all) suggestions are gratefully received.

Cheers
Martin

rossdonald97, 2020-12-02:
Thanks. Sounds good. Saw the HTTPS change was made and the variable fixed. Will keep the issue up and can close when the 1 week of history json link works.

llvllch, 2020-12-02:
URL is working. 

For now, the fallback is there to prevent the CoinApi limits becoming an issue. If someone sat pressing refresh until they got a temporary block, the script will go to a cached version of historic data and a live Coinbase value. 

I think this is close enough to a solution to close the issue. Do you agree?

rossdonald97, 2020-12-03:
Works for me :)

closed 2020-12-03 by rossdonald97