veebch/btcticker #4: Unable to run receive error

skydiveguy, 2020-12-03:
Followed all the steps on your githum page. Installed software as directed from Waveshare. Test files from Waveshare display perfectly.

When attempting to run "pyton3 btcticker.py" an error is thrown....

Traceback (most recent call last):
  File "btcticker.py", Line 9, in < module >
    from waveshare_epd import epd2in7
ModuleNotFoundError: No module named 'waveshare_epd'

llvllch, 2020-12-03:
The waveshare_epd must not be installed. The waveshare test ones work because the module files are in the directory structure.

You need to run the setup script for the waveshare module to be installed.
```
e-Paper/RaspberryPi&JetsonNano/python/setup.py install
```

skydiveguy, 2020-12-04:
Negative.

Now I am getting...

Error: error in 'egg_base' option: 'Lib' does not exist or is not a directory

Of course the pip list command does not list the module because it didnt install from the first line.

llvllch, 2020-12-04:
I've just (on a clean Rasbian build) ran

```
cd e-Paper/RaspberryPi&JetsonNano/python
sudo python3 setup.py install
```
and it installed the module.

skydiveguy, 2020-12-04:
Working nowThank you... very strange... I navigated to that same directory before replying but it failed with the message i gave.
Now its working.
Thanks again.

llvllch, 2020-12-04:
Great, I've updated the README to explicitly mention that step. Thanks for finding that wrinkle.

closed 2020-12-04 by llvllch