veebch/btcticker #5: Percentage does not display decimal

rossdonald97, 2020-12-06:
Hi,

Currently the +/-  change percentage shows a whole number (aka 5% instead of 5.25%), in the script the rounding attempts to round to two decimal places so I'm not sure if showing just a whole int is intentional?

Example line currently in use: 
`draw.text((110,95),str("%+d" % round((pricestack[-1]-pricestack[1])/pricestack[-1]*100,2))+"%",font =font_date,fill = 0)`

The reason 5% is showing and not, say 5.25% is the "%+d" formatting on the draw text lines. The rounding function is to 2 places, but that will only show a whole int. A floating point could instead be used (%f) but then it'll push to like 5 decimals as default. This can be manually set to 2 using "%+.2f", where 2 represents the number of decimal points wanted.

Made the change locally but didn't want to fork and make a pull req since I wasn't sure if it was intended functionality to limit to a whole num.
Opinions? 😃

llvllch, 2020-12-06:
Hey Ross,

Intentional! Seeing as one percent is already two orders of magnitude smaller than the price, I thought rounding to whole numbers is fine and more legible in that tiny font.

rossdonald97, 2020-12-06:
Understandable! Closed.

closed 2020-12-06 by rossdonald97

llvllch, 2020-12-06:
Although looking at it, I feel like pricestack[1] should be pricestack[0]