veebch/btcticker #6: Ran into issues with attempting to display coins with value less than 1.

snarflakes, 2020-12-07:
I think any coin with a value of a single digit, 1-9, should also include 2 decimal places.  any coin with a value of less than 1 should have at least 2 decimal places but ideally 3.  As currently coded doesn't seem to allow for display of value less than 1.  Maybe I did it wrong? Feel free to run snarflakes fork to see what happens. The 7day movement matches what coingecko shows. It just displays $0 as the value. 

So happy to have found your code, great project for me to explore.

llvllch, 2020-12-07:
Thanks!

I've stopped it from rounding the price to an integer and told it to give 5 significant figures instead

float('%.5g' % pricenow)

See if that helps

snarflakes, 2020-12-07:
That works perfectly. I just tested it with a coin: $0.025817.  I will also try to test it with a single digit number 1-9.xx
Many thanks

closed 2020-12-07 by snarflakes