veebch/btcticker #8: INFO:root:[Errno 13] Permission denied: '/btcticker/images/spark.png'

kevinbright, 2020-12-10:
I'm the guy from Reddit that was having troubling getting the script to start on reboot.  I decided to try it all over again with a fresh install on a new SD card and now I cannot run the script.  When I try: 

`python3 btcticker.py`

I get 

`INFO:root:[Errno 13] Permission denied: '/btcticker/images/spark.png'`

Any ideas?

kevinbright, 2020-12-10:
I should also add that when trying to clone the repository using `git clone https://github.com/llvllch/btcticker.git,` I got: 

`fatal: could not create work tree dir 'btcticker': Permission denied`

So I used `sudo git clone https://github.com/llvllch/btcticker.git` and I was able to clone it.  Not sure if that is part of the problem.

sncsoft, 2020-12-10:
It looks like you are cloning a github repository to the root of your file
system as a root user. You have to clone it in the folder under another
user, like pi. Login as pi (or whatever username you use), create working
folder (I like Projects) -
mkdir ~/Projects
cd ~/Projects
git clone https://github.com/llvllch/btcticker.git
cd btcticker
follow the btcticker installation instructions

kevinbright, 2020-12-10:
You are correct.  I am an idiot.  Feel free to close.  Thanks.

closed 2020-12-10 by llvllch