veebch/btcticker #9: error after approximately 6 hours

jmdeacon, 2020-12-12:
I am using a raspberry pi zero w with the 2.7 screen just as you describe. It runs fine for around 6 hours and then stops working.

The error keeps happening.

Here is a screenshot of the error message.

[image]

llvllch, 2020-12-12:
That looks like coingekco close the connection. I've not experienced that. Did you change the refresh rate?

I just broadened out the cases where that script falls back to coinapi. Try pulling the latest source and seeing how that behaves after 6h.

jmdeacon, 2020-12-13:
When I first built my ticker, I had changed the config file to refresh every 60 seconds instead of the default 600 seconds. After reading your reply, I decided to switch it back to 600. Now it has been up and running for nearly 24 hours.

Therefore, it appears that the error occurred from trying to refresh every minute. I'm fine with keeping it at every 10 minutes but if anyone else has this problem, you can suggest they go back to every 10 minutes.

llvllch, 2020-12-13:
Great!
I imagine it was the coingecko API.... the change I made to the code should now revert to coinapi in that case. Closing the issue for now.

closed 2020-12-13 by llvllch

tmlee, 2020-12-14:
Hey from CoinGecko here, surprised to see how there can be issue if you are only hitting the API once every minute; unless you are somehow making unusually more calls which may be lead to a rate limit problem.

llvllch, 2020-12-14:
It sounds more likely to be an intermittent internet connection. The code currently checks for internet when it starts and then assumes that it is there after that. 

@tmlee Out of interest....What is the daily limit for api calls to coingekco?

tmlee, 2020-12-15:
> It sounds more likely to be an intermittent internet connection. The code currently checks for internet when it starts and then assumes that it is there after that.
> 
> @tmlee Out of interest....What is the daily limit for api calls to coingekco?

Its currently a minutely limit of 100 requests, but my suggestion is to keep it at 60 calls or less per minute to be absolute safe due to spikes in the public API

llvllch, 2020-12-15:
thanks! The ePaper takes a few seconds to refresh, so the code couldn't exceed that if it tried.

llvllch, 2021-01-04:
> Hey from CoinGecko here, surprised to see how there can be issue if you are only hitting the API once every minute; unless you are somehow making unusually more calls which may be lead to a rate limit problem.

@tmlee Hey.... quick question: Does the coingecko have a way of pulling the price for a specific exchange?