veebch/btcticker #10: wifi disconnect

snarflakes, 2020-12-16:
I think my wifi shut down randomly, and then it causes btcticker application to shut down because there isn't code to allow it to restart if wifi is lost?

llvllch, 2020-12-16:
Hey :)
It could be that it is the slightly older version that throws an error when the internet goes down. The current version *should* just wait. Pull the latest version and see if it solves the problem.

jmdeacon, 2020-12-16:
I might be having the same or similar problem. Running the latest version.

`INFO:root:Got Historical Data For Last Week from CoinGecko
DEBUG:matplotlib.axes._base:update_title_pos
DEBUG:matplotlib.axes._base:update_title_pos
DEBUG:PIL.PngImagePlugin:STREAM b'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM b'sBIT' 41 4
DEBUG:PIL.PngImagePlugin:b'sBIT' 41 4 (unknown)
DEBUG:PIL.PngImagePlugin:STREAM b'pHYs' 57 9
DEBUG:PIL.PngImagePlugin:STREAM b'tEXt' 78 57
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 147 2125
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
Traceback (most recent call last):
  File "btcticker.py", line 35, in internet
    socket.socket(socket.AF_INET, socket.SOCK_STREAM).connect((host, port))
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "btcticker.py", line 325, in <module>
    main()
  File "btcticker.py", line 253, in main
    if internet():
  File "btcticker.py", line 38, in internet
    logging.info("No internet:"+ex)
TypeError: can only concatenate str (not "timeout") to str`

llvllch, 2020-12-16:
Ah.... there has been a change since then ```logging.info("No internet:"+ex)``` is complaining that ex isn't a string.

jmdeacon, 2020-12-16:
Ok, just pulled.

snarflakes, 2020-12-16:
I will pull and recheck! thanks

llvllch, 2020-12-16:
I'll not close until you confirm that it is untroubled by network blips

jmdeacon, 2020-12-16:
It is no longer crashing for me during internet blips.

llvllch, 2020-12-21:
Closing for now..... all is quiet on the netblip front

closed 2020-12-21 by llvllch