veebch/btcticker #11: Time out of sync

snarflakes, 2020-12-17:
Time seems to get disconnected (unsync) with a basic wifi blip (i did not do a full router disconnect) i think. This time I had a wifi blip the coin price resumed updating normally, but the time got stuck.  Even if changing coins, which are still updating normally, time is staying stuck seeming to mark the moment of the wifi blip.

I will have to try and run this on your code (this occurred on my fork).

snarflakes, 2020-12-17:
i will close this until i can duplicate it again.

closed 2020-12-17 by snarflakes

llvllch, 2020-12-17:
OK.... I vaguely recall that the Pi doesn't have a clock so until it can
see the internet/ your LAN, the time isn't synced. Others will know better
than me though......