veebch/btcticker #12: Suggestion - Token Image Icons

RattyDAVE, 2020-12-22:
First thanks for the great work!

I have a suggestion to grab the token icon.

`https://www.coingecko.com/api/documentations/v3#/coins/get_coins__id_`

eg.

`https://api.coingecko.com/api/v3/coins/bitcoin?tickers=false&market_data=false&community_data=false&developer_data=false&sparkline=false
`
Then [image][large] and grab the image from the URL.

Regards

llvllch, 2020-12-22:
That is a good idea...I'll look into it

RattyDAVE, 2020-12-22:
I see you are using PIL

So how about... 

```
from PIL import Image

file_in = "test1.png"
img = Image.open(file_in)

file_out = "test1.bmp"
img.save(file_out)
```

llvllch, 2020-12-23:
Well, now I have no excuse not to do it. Thanks :)

llvllch, 2021-01-14:
Done!

closed 2021-01-14 by llvllch