veebch/btcticker #13: Error running: python3 -m pip install -r requirements.txt

mjoerck, 2020-12-22:
When I run the command stated in the title I receive an error when building wheels for collected packages: numpy. 
I get the following error text back:

````
Failed building wheel for numpy
 Running setup.py clean for numpy
 Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-tylsvy0g/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" clean --all:
 Running from numpy source directory.

 `setup.py clean` is not supported, use one of the following instead:

 - `git clean -xdf` (cleans all files)
 - `git clean -Xdf` (cleans all versioned files, doesn't touch
                  files that aren't checked into the git repo)

  Add `--force` to your command to use it anyway if you must (unsupported).
````
Is there a workaround for this, or am I missing something? :)

llvllch, 2020-12-22:
Was this on a fresh install of Rasbian? I've not ran into this one.... hopefully someone who speaks pip better than I do will help.

An alternative is to just install the few dependencies manually

try ```python3 -m pip install numpy``` and see if it still complains

mjoerck, 2020-12-22:
Yes, on a clean install - I will try installing them manually.

mjoerck, 2020-12-22:
Hmm, when I install all the libraries manually and try to run the program using:

`python3 btcticker.py`

I get the following:

```
Traceback (most recent call last):
  File "btcticker.py", line 2, in <module>
    from PIL import Image, ImageOps
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/Image.py", line 94, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory
```

Should I try starting over with fresh install of rasbian again?

llvllch, 2020-12-22:
If that's the quickest thing to do. Some other people have mentioned that they struggled with the Python deps. Maybe some helpful soul will chime in on this thread.

samdogg7, 2020-12-22:
I am having the same issue on a fresh install. I will update if I find a solution.

llvllch, 2020-12-22:
Try on the latest version. Pipreqs seems to have made a bit of a mess. I did a manual rebuild and test

mjoerck, 2020-12-22:
> Try on the latest version. Pipreqs seems to have made a bit of a mess. I did a manual rebuild and test

I did a full reset of the pi and ran:

`python3 -m pip install -r requirements.txt`

without any errors; however, I still get the traceback to line 2... :/

llvllch, 2020-12-22:
Seems odd. Do a ```pip list``` and see what version of Pillow you have

mjoerck, 2020-12-22:
It is running pillow version 6.0.0:

```
pi@raspberrypi:~/btcticker $ pip3 list
Package         Version
--------------- ---------
asn1crypto      0.24.0
certifi         2018.8.24
chardet         3.0.4
cryptography    2.6.1
cycler          0.10.0
entrypoints     0.3
idna            2.6
keyring         17.1.1
keyrings.alt    3.1.1
kiwisolver      1.3.1
matplotlib      3.0.3
nh-currency     1.0.1
numpy           1.16.2
olefile         0.46
Pillow          6.0.0
pip             18.1
pycrypto        2.6.1
PyGObject       3.30.4
pyparsing       2.4.7
python-apt      1.8.4.1
python-dateutil 2.8.1
pyxdg           0.25
PyYAML          5.3.1
requests        2.21.0
RPi.GPIO        0.7.0
SecretStorage   2.3.1
setuptools      40.8.0
six             1.12.0
spidev          3.5
ssh-import-id   5.7
urllib3         1.24.1
waveshare-epd   0.0.0
wheel           0.32.3
```

llvllch, 2020-12-22:
try ```sudo apt-get install libopenjp2-7```

mjoerck, 2020-12-22:
> try `sudo apt-get install libopenjp2-7`

Hooray! It works now, thank you very much!

llvllch, 2020-12-22:
Not too hard, you just need to find an api

mjoerck, 2020-12-22:
> Not too hard, you just need to find an api

Thanks! I'll take a look at it tomorrow. Yahoo has a pretty nice API if you decide to take a look at it yourself. ;) Should I close the issue? I don't really know how this works :P

llvllch, 2020-12-22:
Yeah... I'll close! Good luck with the api.... if it works, don't forget to share :)

closed 2020-12-22 by llvllch