veebch/btcticker #14: Install and run at boot.

RattyDAVE, 2020-12-25:
The instructions to Install from a new image install and get to run at boot.

```
sudo raspi-config
#Choose Interfacing Options -> SPI -> Yes  to enable SPI interface
sudo reboot
```

Wait for the reboot!

Login and then get root access

```
sudo bash
```

And everything below is done as root

```
mkdir ~/Projects
mkdir ~/Projects/waveshare
cd ~/Projects/waveshare

wget http://www.airspayce.com/mikem/bcm2835/bcm2835-1.60.tar.gz
tar zxvf bcm2835-1.60.tar.gz 
cd bcm2835-1.60/
./configure
make
make check
make install
#For more details, please refer to http://www.airspayce.com/mikem/bcm2835/

cd ~/Projects/waveshare
git clone https://github.com/waveshare/e-Paper
cd e-Paper/RaspberryPi_JetsonNano/python
python3 setup.py install

cd ~/Projects
git clone https://github.com/llvllch/btcticker.git
cd btcticker

python3 -m pip install -r requirements.txt

vi /etc/rc.local 

## Add the following before the last line (exit 0)

/usr/bin/python3 ~/Projects/btcticker/btcticker.py &
```
When you reboot next btcticker will start up.

RattyDAVE, 2020-12-25:
[image]

(you may notice I have 1 DAY)

Frankenmint, 2021-01-01:
FYI I got the rc.local to run on boot this way (using the pi user rather than installing things via root):

`sudo -u pi /usr/bin/python3 /home/pi/btcticker/btcticker.py &`

trying to run the straight script gave me trouble as you could imagine because the dependencies installed on the user since I used default directions for setting up raspian lite

llvllch, 2021-01-01:
@RattyDAVE Looks great (If you pull that green tab the screen protector comes off)

closed 2021-01-01 by llvllch