veebch/btcticker #16: Trying to replace functionality of button 4, cannot use it to clear screen

elric25, 2020-12-27:
My use-case means I only have one fiat currency to worry about, so I wanted to use the fourth button to wipe the display clean and stop the program. I took some material from elsewhere in the code as well as some of the e-paper example code provided by the manufacturer, but it doesn't seem to be doing anything

'if key4state == False:
epd = epd2in7.EPD()
epd.Clear(0xFF)
epd.Sleep()
time.sleep(3)
epd2in7.epdconfig.module_exit()
exit()'

I can definitely tell that this does stop the program as the display no longer updates and none of the other buttons work, but the screen stays the same with whatever was displayed at the time.

llvllch, 2020-12-27:
It's hard to without seeing your code, but the behaviour you describe is
consistent with an  error crashing the code.

Is the code indented correctly?

llvllch, 2020-12-27:
It's hard to without seeing your code, but the behaviour you describe is
consistent with an  error crashing the code.

Is the code indented correctly?

`if key4state == False:
    epd = epd2in7.EPD()
    epd.Clear(0xFF)
    epd.Sleep()
    time.sleep(3)
    epd2in7.epdconfig.module_exit()
    exit()`

elric25, 2020-12-27:
Yeah, the indents are all correct I just didn't copy it in here correctly. When I hit the fourth button I'm not seeing any output in the console, is there another place I might see some information that could help?

llvllch, 2020-12-27:
After you crash it, run from the terminal `python3 btcticker.py` and watch the logging output

elric25, 2020-12-27:
Okay so whenever the screen settles or updates it puts out two debug lines of e-Paper busy and then e-Paper busy release, when I hit the fourth button all I get is another e-Paper busy. 

I'm now testing the clearing code with the ctrl + c shutdown command in terminal, still doesn't work and I need to hit ctrl + c twice to actually get out of the logging.

llvllch, 2020-12-27:
My guess (and it is a guess) is that you don't initialise the screen before you send the clear command, so it doesn't clear, but you do have an exit() statement which ends the script. 
Try ditching the 
`    
epd2in7.epdconfig.module_exit()
exit()` and popping an
`epd.Init_4Gray()` before the clear statement.

closed 2020-12-27 by llvllch