veebch/btcticker #17: Added Auto Scroll Function

Derpalupagus, 2020-12-31:
Added an auto scrolling function on top of the Button 1 key press - this will automatically scroll through all coins configured in config.yaml without having to press the button.  Created a new config.yaml file to add a separate scrolling frequency (in seconds).

btcticker.py:
[btcticker.py.txt](https://github.com/llvllch/btcticker/files/5756573/btcticker.py.txt)

config.yaml:
[config.yaml.txt](https://github.com/llvllch/btcticker/files/5756572/config.yaml.txt)

Also, I've found it much easier to install and configure as pi instead of root, and run with cron instead of using rc.local.  This way you can add your own coin icons/images to the /btcticker/images/currency folder without exposing root:

As user pi:

crontab -e

//Add this line at the bottom of the file:

@reboot sudo python3 <folder that btcticker.py is in> &

//Save the file (Crtl-O and Ctrl-X) and reboot

llvllch, 2020-12-31:
Brilliant..... this was a something I'd considered adding as an additional mod. 

One comment: Is your crontab entry missing the path to the script?

Derpalupagus, 2020-12-31:
Weird, I don't know why the crontab entry didn't show above:

@reboot sudo python3 /"path to wherever btcticker.py is"/btcticker.py &

llvllch, 2021-02-10:
I've added a scroll capability... thanks!

closed 2021-02-10 by llvllch