veebch/btcticker #19: Crashes with Out of memory error after a few days

DigiDr, 2021-01-03:
After a few days it will crash out:

```
Traceback (most recent call last):
  File "btcticker.py", line 325, in <module>
    main()
  File "btcticker.py", line 308, in main
    makeSpark(pricestack)
  File "btcticker.py", line 123, in makeSpark
    plt.savefig(os.path.join(picdir,'spark.png'), dpi=17)
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/pyplot.py", line 690, in savefig
    fig.canvas.draw_idle()   # need this if 'transparent=True' to reset colors
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/backend_bases.py", line 1899, in draw_idle
    self.draw(*args, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/backends/backend_agg.py", line 396, in draw
    self.renderer = self.get_renderer(cleared=True)
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/backends/backend_agg.py", line 417, in get_renderer
    self.renderer = RendererAgg(w, h, self.figure.dpi)
  File "/home/pi/.local/lib/python3.7/site-packages/matplotlib/backends/backend_agg.py", line 87, in __init__
    self._renderer = _RendererAgg(int(width), int(height), dpi)
MemoryError: In RendererAgg: Out of memory
```

Great project. Thanks!

llvllch, 2021-01-03:
Hey, good catch. I've not had this one, but I think it is probably good practice to close the plot in the loop. I'll make that change.

Thanks!

llvllch, 2021-01-03:
Let me know if the change stops this happening :)

DigiDr, 2021-01-03:
Thanks, will report back.

Derpalupagus, 2021-01-04:
Fixed, see my post here:  https://github.com/llvllch/btcticker/issues/20

llvllch, 2021-01-04:
Fixed!

closed 2021-01-04 by llvllch