veebch/btcticker #20: Fixed crashing problem, fixed connection issue, cleaned up auto scroll feature, and fixed number display

Derpalupagus, 2021-01-04:
1. Fixed the problem with the script being killed on high memory usage because the plot function was not being cleared from memory after displaying the chart. 
2. Cleaned up the Auto Scroll feature logic - got rid of updatefrequency since we're now using scrollfrequency to grab data
3. Got rid of the script kill when Coin Gecko is unavailable or no internet.  Now the script will keep trying indefinitely until the connection is good and will not exit the script.
4. Fixed the number display issue; will now display with 2 decimal places unless the value is > 99,999.  Since Coin Gecko returns integers above 10,000, any price > 10,000 will always have zeroes after the decimal.  Prices less than 10,000 will show numbers after the decimal.

This version is stable, been running on a Pi 4 Model B 2GB for a few days now with no problems. Use these these files; rename to btcticker.py and config.yaml and FTP them onto your Raspberry Pi:

[btcticker.py.txt](https://github.com/llvllch/btcticker/files/5763253/btcticker.py.txt)
[config.yaml.txt](https://github.com/llvllch/btcticker/files/5763254/config.yaml.txt)

https://user-images.githubusercontent.com/45051328/103507169-c8825400-4e1b-11eb-9042-ccfaa97d18ee.mp4

Enjoy!

llvllch, 2021-01-04:
Impressive! I will weave these back into the code on the repo. I'd already made some of the changes but I'll take a look at how you handled them to, as there is a good chance you did it better!

Sidenote: That's a fine looking case.

Seeing as this is running stable, I'll close this issue for now. 
Again, thanks!

closed 2021-01-04 by llvllch

Derpalupagus, 2021-01-04:
Thanks for the feedback - The case will be up on Thingiverse here https://www.thingiverse.com/thing:4709963 in 24 hours, I guess I have to wait a day before it will publish.

Great project!

llvllch, 2021-01-04:
Thanks! I could add the thingiverse link to the frames directory, once it goes live. Is that ok?

Derpalupagus, 2021-01-04:
Absolutely - I also linked this project to the Thing file; sorry I should have asked first.  I can remove it if you want me to.

llvllch, 2021-01-04:
No problem at all!

sterdeus, 2021-05-24:
@Derpalupagus Beautiful case. The link to thingiverse is broken. If there any way you share it again? Thank you!