veebch/btcticker #21: line 216 in btcticker.py

SBSchram, 2021-01-06:
should be time.sleep(10)

llvllch, 2021-01-06:
Thanks!

Too many syntaxes!

closed 2021-01-06 by llvllch

SBSchram, 2021-01-07:
Hi Martin,

I'm just learning Python. Lots of interesting syntax, countless libraries.
I learn a lot by looking at code, running it, modifying it, and then
crashing my system repeatedly. Thank you for providing me with a lot of fun!

Steve

Steven Schram PhD, DC, LAc
140 East 28th St (Suite 1F)  New York, NY 10016 - 212-696-4426
http://www.HandyWorks.com

HIPAA Notification: *This electronic message contains information that may
be privileged, confidential, otherwise protected from disclosure. The
information herein is intended to be for the addressee only. If you are not
the intended addressee, any disclosure, copying, distribution or use of the
contents of this message (including* *attachments) is prohibited. If you
have received this electronic message in error, please notify the sender
immediately and destroy the original message and all copies.  *