veebch/btcticker #22: e-paper busy

15rhema, 2021-01-06:
Hi, I followed all the instructions a couple times and I'm getting this line when running btcticker.py: 

`root@raspberrypi:~/Projects/btcticker# python3 btcticker.py
INFO:root:epd2in7 BTC Frame
INFO:root:{'ticker': {'currency': 'bitcoin,litecoin,bitcoin-cash,ethereum,stellar,monero, uniswap', 'fiatcurrency': 'gbp,usd,eur,jpy,chf', 'updatefrequency': 600}, 'display': {'orientation': 90, 'inverted': True}}
INFO:root:['bitcoin', 'litecoin', 'bitcoin-cash', 'ethereum', 'stellar', 'monero', 'uniswap']
INFO:root:['gbp', 'usd', 'eur', 'jpy', 'chf']
INFO:root:bitcoin
INFO:root:gbp
INFO:root:Getting Data
INFO:root:https://api.coingecko.com/api/v3/coins/markets?vs_currency=gbp&ids=bitcoin
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.coingecko.com:443
DEBUG:urllib3.connectionpool:https://api.coingecko.com:443 "GET /api/v3/coins/markets?vs_currency=gbp&ids=bitcoin HTTP/1.1" 200 None
INFO:root:Got Live Data From CoinGecko
INFO:root:https://api.coingecko.com/api/v3/coins/bitcoin/market_chart/range?vs_currency=gbp&from=1609368509&to=1609973309
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.coingecko.com:443
DEBUG:urllib3.connectionpool:https://api.coingecko.com:443 "GET /api/v3/coins/bitcoin/market_chart/range?vs_currency=gbp&from=1609368509&to=1609973309 HTTP/1.1" 200 None
INFO:root:Got Historical Data For Last Week from CoinGecko
DEBUG:matplotlib.axes._base:update_title_pos
DEBUG:matplotlib.axes._base:update_title_pos
DEBUG:PIL.PngImagePlugin:STREAM b'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM b'sBIT' 41 4
DEBUG:PIL.PngImagePlugin:b'sBIT' 41 4 (unknown)
DEBUG:PIL.PngImagePlugin:STREAM b'pHYs' 57 9
DEBUG:PIL.PngImagePlugin:STREAM b'tEXt' 78 57
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 147 2529
DEBUG:root:e-Paper busy
`

I did some googling around and it's possible my e-paper is bad? I also tried another fix someone posted by re-compiling the python code for the e-paper but no dice.

llvllch, 2021-01-06:
Does it just hang at that? This is the tail end on of the log on mine after a successful refresh. The 4th line down seems to differ..... 

Pretty sure this is a waveshare thing. All the logs for the script (the ones that start with INFO) seem happy...

--snip----

DEBUG:PIL.PngImagePlugin:b'sBIT' 41 4 (unknown)
DEBUG:PIL.PngImagePlugin:STREAM b'pHYs' 57 9
DEBUG:PIL.PngImagePlugin:STREAM b'tEXt' 78 57
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 147 1928
 DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release

15rhema, 2021-01-06:
Yes, it just stays there forever with nothing displayed on the screen. After I changed the python code it displayed this:

DEBUG:root:Forced e-Paper busy release
DEBUG:root:e-Paper busy release
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
DEBUG:root:Forced e-Paper busy release
DEBUG:root:e-Paper busy release

But I still never get anything on the display.

llvllch, 2021-01-06:
What about the test codes in the example directory on the e-Paper stuff? 

I'm of little use on this..... definitely an epaper issue, but I wouldn't know if it is hardware.

closed 2021-01-07 by llvllch

llvllch, 2021-01-08:
Something that did occur to me: 

Is the screen connection ok? Are the pins in the hat correctly?

15rhema, 2021-01-08:
I haven't had time the last few days to work on this more, but I'm working on testing some demo code on the display to see if I can get that working. Here is a video of my connection. 

https://user-images.githubusercontent.com/5685847/104044846-9281fe80-5192-11eb-8cfb-cb5eb9466b34.mp4

llvllch, 2021-01-08:
That looks fine to me. I occasionally misalign the pins, and that has cause the behaviour you describe. 

(Nice case)

15rhema, 2021-01-08:
Ok, thanks for the sanity check. I'll have to keep messing around and see if I can get the demo to work. Worst case scenario I'm still in the return window for the e-paper so I could just return and try another one if this one turns out to be bad.

Thanks, the case is still a prototype, it's a remix of another design that I found on thingiverse.  I still need to update the button location.

15rhema, 2021-01-08:
Confirmed that the demo code does the same thing. I guess I'm going to return it. Must be a bad unit. Thanks for your help!