veebch/btcticker #23: Random crashes - exception

snowboarder04, 2021-01-09:
`DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
INFO:root:No internet
INFO:root:No internet
INFO:root:No internet
INFO:root:No internet
INFO:root:No internet
INFO:root:No internet
INFO:root:Getting Data
INFO:root:https://api.coingecko.com/api/v3/coins/markets?vs_currency=usd&ids=ethereum
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.coingecko.com:443
DEBUG:urllib3.connectionpool:https://api.coingecko.com:443 "GET /api/v3/coins/markets?vs_currency=usd&ids=ethereum HTTP/1.1" 200 None
INFO:root:Got Live Data From CoinGecko
INFO:root:https://api.coingecko.com/api/v3/coins/ethereum/market_chart/range?vs_currency=usd&from=1609974311&to=1610233511
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.coingecko.com:443
DEBUG:urllib3.connectionpool:https://api.coingecko.com:443 "GET /api/v3/coins/ethereum/market_chart/range?vs_currency=usd&from=1609974311&to=1610233511 HTTP/1.1" 503 None
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:spi end
DEBUG:root:close 5V, Module enters 0 power consumption ...
Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 208, in fullupdate
    pricestack=getData(config,CURRENCY,FIAT)
  File "/home/pi/btcticker/btcticker.py", line 74, in getData
    rawtimeseries = requests.get(geckourlhistorical).json()
  File "/usr/lib/python3/dist-packages/requests/models.py", line 897, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 316, in <module>
    main()
  File "/home/pi/btcticker/btcticker.py", line 302, in main
    lastcoinfetch=fullupdate()
  File "/home/pi/btcticker/btcticker.py", line 216, in fullupdate
    sleep(10)
NameError: name 'sleep' is not defined`

Config:

`display:  
  inverted: false  
  orientation: 180  
ticker:  
  currency: ethereum,bitcoin,bitcoin-cash  
  exchange: default  
  fiatcurrency: usd,gbp  
  sparklinedays: 3  
  updatefrequency: 60`

Note: Those `INFO:root:No internet` messages are false, the device does have internet

llvllch, 2021-01-10:
Hey Rob.... the first thing to try is using the latest version, This is
from memory because the code isn't in front of me, but line 216 does not
say 'sleep(10)' in the latest version.

The error aboit

llvllch, 2021-01-10:
Yep..... you're on an older version. Do a `git pull` from the directory the script is in and try again

snowboarder04, 2021-01-10:
Thanks, will give that a go

snowboarder04, 2021-01-10:
Since pulling the latest it's been stable. I reckon I must've typo'd somewhere while reviewing the source. I've closed this one out and thanks for the great project!

closed 2021-01-10 by snowboarder04