veebch/btcticker #24: ModuleNotFoundError: No module named 'currency'

forg0t, 2021-01-11:
Traceback (most recent call last):
  File "btcticker.py", line 5, in <module>
    import currency
ModuleNotFoundError: No module named 'currency'

I've tried rerunning the requirements and even ` python3 -m pip install pip nh-currency` but the requirements are already satisfied.  It was working fine for half a day and then stopped updating.  I even did a clean install, any ideas?

llvllch, 2021-01-11:
do you see it when you do `python3 -m pip list`?

forg0t, 2021-01-11:
I do, I see nh-currency 1.0.1

llvllch, 2021-01-11:
Excellent...... next question: are you definitely running the script with
python3?

forg0t, 2021-01-11:
Embarassing.  I've been running `sudo python3 btcticker.py` instead of `python3 btcticker.py`

Sorry to waste your time, it works now.

closed 2021-01-11 by forg0t

uududde, 2022-02-05:
> Embarassing. I've been running `sudo python3 btcticker.py` instead of `python3 btcticker.py`
> 
> Sorry to waste your time, it works now.

Sorry to hang onto this... but I'm having the same error message. 
If I use it without sudo I receive: 
```
pi@raspberrypi:~/btcticker` $ python3 ./btcticker.py 
Illegal instruction
```
if I run it with sudo I get: 
```
pi@raspberrypi:~/btcticker $ sudo python3 ./btcticker.py 
Traceback (most recent call last):
  File "./btcticker.py", line 5, in <module>
    import currency
ModuleNotFoundError: No module named 'currency'
```
_python3 -m pip list_  gives both _Currencey_ and _nh-currency_ installed.

veebch, 2022-02-05:
You want to run it **without** using sudo, that means that as a user (pi I assume) you need to install the required modules for the user.

Try running:
          python3 -m pip install -r requirements.txt

as a user (ie taking care not to use sudo). That should add the required modules, (including nh-currency)

uududde, 2022-02-05:
oh ok, thanks! using that without sudo gives me "Requirement already satisfied" everywhere. 
e.g.: "Requirement already satisfied: pillow>=8.3.2 in /home/pi/.local/lib/python3.7/site-packages (from -r requirements.txt (line 4)) (9.0.0)
"
... running gives me the same problem as mentioned before.

veebch, 2022-02-05:
I wonder why it isn't complaining about nh-currency...
try:
      python3 -m pip install nh-currency

uududde, 2022-02-05:
```
$ python3 -m pip install nh-currency
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: nh-currency in ./.local/lib/python3.7/site-packages (1.0.1)
Requirement already satisfied: requests in ./.local/lib/python3.7/site-packages (from nh-currency) (2.25.1)
Requirement already satisfied: idna<3,>=2.5 in /usr/lib/python3/dist-packages (from requests->nh-currency) (2.6)
Requirement already satisfied: chardet<5,>=3.0.2 in /usr/lib/python3/dist-packages (from requests->nh-currency) (3.0.4)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/lib/python3/dist-packages (from requests->nh-currency) (1.24.1)
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests->nh-currency) (2018.8.24)
```
and
```
$ python3 -m pip install --upgrade Pillow
...
Successfully uninstalled Pillow-9.0.0
Successfully installed Pillow-9.0.1
```
 
the code looks like it is working now. it starts, reads the config.yaml and says `INFO:root:Getting Data` ... where it's hanging / taking ages and not moving further... `packet_wrtie_wait` ssh connection failed... next try it worked... suddenly a very nice picture on the eink appears. 
Thank you!