veebch/btcticker #25: "No internet" leading to "Coingecko is unreachable: Exit the script"

snarflakes, 2021-01-11:
Despite updating, I keep getting INFO: No internet x 2, which then eventually reconnects, gets data from coingecko, and then on the next refresh hits "No internet" x 1,  then "getting data," and then at some point during getting data from coingecko leads to "Coingecko is unreachable - Exit the script."  
Problem: During times of weak wifi, or internet blips occuring during the coingecko data retrieval stage, coingecko data loss or interruption leads to shut down of program instead of reattempting.  This has been occuring once every 2 days.  
Solution: Do we need to add in another pause somewhere during the coingecko data grab? Or perhaps a Re-attempt if there was a disruption during coingecko download so the program doesn't shut down when hitting this challenge?

llvllch, 2021-01-11:
The current version should just wait 10 seconds and try again if the
coingecko pull fails. This should also happen if an exception is thrown mid
pull. I'll have another look at the code. I think my use of systemd might
be stopping me from seeing it.

snarflakes, 2021-01-12:
I will pull again, to retest in case I missed an update.

llvllch, 2021-01-13:
If connection drops, a holding screen should display until connection is
re-established. Let me know how you get on….

snarflakes, 2021-01-15:
Thanks,
The problem has changed since the last pull. I’m checking frequency pattern and to make sure it is repeating exactly the same errors. I will update in a few days.

Sent from ProtonMail Mobile

snarflakes, 2021-01-15:
Okay this error repeated x2 now.  Re-occurring every 1-2 days. Since I pulled your last changes per my previous comment this is occurring.  The old problem which said Coingecko is unreachable - Exit the script doesn't seem to be occurring any more.

This is what is happening now:

I see one: INFO:root: No internet
it moves forward no issues and getting data
Then it goes 

DEBUG:root:spi end
DEBUG:root: close 5V, module enteres 0 power consumptions...
Trace back (most recent call last):
File "btc ticker.pay", line 316 n <module>
Main()
File "btc ticker.pay", line 271, in main
Key1state = GPIO.input(key1)
RuntimeError: please set pin numbering mode using GPIO.setmode(GPIO.BOARD or GPIO.setmode(GPIO.BCM)

The pi screen displays your little Axie, when this has happened. Screen still detached and accessible.

Is a reboot occurring? I haven't seen the error happen live, I just find the pi in this state.

llvllch, 2021-01-15:
what is File “btc ticker.pay”? Or is that just some typos that appears
during cut and paste. It looks like pressing key 4 is throwing that error,
I’ll try to reproduce.

snarflakes, 2021-01-15:
sorry autocorrect.  Should read .py

llvllch, 2021-02-10:
Closing for now. No news is good news :)

closed 2021-02-10 by llvllch

SR00000, 2021-02-20:
I wanted to comment on this because I had this problem earlier and found the issue. My internet gateway had "advanced security" enabled and it automatically blocked outgoing pings from my Pi because it thought it was trying to DDOS another network. You have to disable that advanced/smart security you may have enabled on your modem/gateway.

If you're able to use the command "curl -X GET https://api.coingecko.com/api/v3/exchanges" and get a screen full of text, that means its definitely been disabled from your modems end. Good luck!