veebch/btcticker #26: Crash with GPIO error

DigiDr, 2021-01-11:
```
DEBUG:root:e-Paper busy release
INFO:root:Getting Data
INFO:root:https://api.coingecko.com/api/v3/coins/markets?vs_currency=usd&ids=bitcoin
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.coingecko.com:443
DEBUG:urllib3.connectionpool:https://api.coingecko.com:443 "GET /api/v3/coins/markets?vs_currency=usd&ids=bitcoin HTTP/1.1" 200 None
INFO:root:Got Live Data From CoinGecko
INFO:root:https://api.coingecko.com/api/v3/coins/bitcoin/market_chart/range?vs_currency=usd&from=1609945864&to=1610205064
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.coingecko.com:443
DEBUG:urllib3.connectionpool:https://api.coingecko.com:443 "GET /api/v3/coins/bitcoin/market_chart/range?vs_currency=usd&from=1609945864&to=1610205064 HTTP/1.1" 200 None
INFO:root:Got price for the last 3 days from CoinGecko
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:spi end
DEBUG:root:close 5V, Module enters 0 power consumption ...
Traceback (most recent call last):
  File "btcticker.py", line 316, in <module>
    main()
  File "btcticker.py", line 271, in main
    key1state = GPIO.input(key1)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
```

DigiDr, 2021-01-11:
To be clear, this happened randomly and without any input from me at the time.

llvllch, 2021-01-11:
I've not seen that one before. Looks like the GPIO module doesn't like that command. Could it be your version of the module?

DigiDr, 2021-01-11:
I don't think so, it's been working perfectly up and until now. Perhaps there is a bug at a lower level?

llvllch, 2021-01-11:
It;s sufficiently odd that I'd do a clean install... that said, someone more knowledgeable may chip in with an actual solution :)

DigiDr, 2021-01-13:
Had this again today quite randomly after 48 hours of everything running smoothly. Hmm

llvllch, 2021-01-13:
same thing in the logs?

DigiDr, 2021-01-13:
Yes

llvllch, 2021-01-13:
The lines
    DEBUG:root:spi end
    DEBUG:root:close 5V, Module enters 0 power consumption ...
Aren't being triggered by this script. That's the waveshare module. I'd see if there is anything about spontaneous sleep on their documentation. The other thing I'd try is another power supply for the pi. In my experience, weird problems almost always come down to cables and/or power

llvllch, 2021-01-24:
Closing for now. I've not been able to reproduce the error.

closed 2021-01-24 by llvllch