veebch/btcticker #31: raise exception( 'could not determine jetson model' )

SR00000, 2021-02-19:
I get this error whether I install btcticker on the home directory or the e-paper directory. Any assistance would be appreciated!

[image]
[image]

llvllch, 2021-02-19:
Hey there….

This isn’t the code throwing the error. It looks like your GPIO module is
complaining. Try updating it maybe?

Good luck!
Martin

SR00000, 2021-02-19:
Hello!

I ran the following updates already:
    sudo apt-get update
    sudo apt-get install python3-pip
    sudo apt-get install python3-pil
    sudo apt-get install python3-numpy
    sudo pip3 install Jetson.GPIO

It results in that despite having updated all of the above. Is there a preference for where btcticker should be cloned/installed to?

llvllch, 2021-02-19:
The prompt makes me think it's a Raspberry Pi and you've installed GPIO for a Jetson board.

SR00000, 2021-02-19:
Im using all the same hardware in the btcticker tutorial (2.7in e-Paper HAT using the Waveshare_epd python module), and I followed the tutorial exactly step by step, so I'm not sure how I installed the GPIO for a different board. Is there an obvious step that wasn't listed in the tutorial because an experienced person would know to do it?

llvllch, 2021-02-19:
sudo pip3 install Jetson.GPIO

isn't in the tutorial is it?

SR00000, 2021-02-19:
Yeah at the earlier steps it says "Install the Waveshare Python module following the instructions on their Wiki." and on the wiki it provides that pip3 install command. If you believe that was done in error I can try to uninstall it?

Thanks in advance for responding, been going back and forth with friends for days trying to figure this out.

llvllch, 2021-02-19:
#python3
    sudo apt-get update
    sudo apt-get install python3-pip
   sudo apt-get install python3-pil
   sudo apt-get install python3-numpy
   sudo pip3 install RPi.GPIO
   sudo pip3 install spidev

SR00000, 2021-02-19:
Ok I will uninstall everything and start from the beginning using only the Pi steps and I'll report back here.

llvllch, 2021-02-19:
sudo pip3 uninstall Jetson.GPIO

might work, or maybe just installing the RPi one will overwrite it

SR00000, 2021-02-19:
That worked! Thank you for your help!

llvllch, 2021-02-21:
Thats great! Closing....

closed 2021-02-21 by llvllch