veebch/btcticker #33: GPIO Error

manuelautomatisch, 2021-02-27:
Traceback (most recent call last):
File "btcticker.py", line 362, in <module>
main()
File "btcticker.py", line 316, in main
key1state = GPIO.input(key1)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)

dkim777, 2021-02-28:
Getting exact error message on new one I just installed.

I wasn't getting this error on one I built two weeks ago though.
only difference is SD card size  32GB vs 12GB(doesn't work)

llvllch, 2021-02-28:
Looks like the GPIO module complaining. Is the script not working or is it complaining when closed?

I can't reproduce the error. What does your
      python3 -m pip list
show for the module RPi.GPIO?

manuelautomatisch, 2021-02-28:
Its also a new error for me. 

Thats the output from **python3 -m pip list**

Package           Version
----------------- -----------
asn1crypto        0.24.0
astroid           2.1.0
asttokens         1.1.13
automationhat     0.2.0
beautifulsoup4    4.7.1
blinker           1.4
blinkt            0.1.2
buttonshim        0.0.2
Cap1xxx           0.1.3
certifi           2018.8.24
chardet           3.0.4
Click             7.0
colorama          0.3.7
colorzero         1.1
cookies           2.2.1
cryptography      2.6.1
cupshelpers       1.0
cycler            0.10.0
docutils          0.14
drumhat           0.1.0
entrypoints       0.3
envirophat        1.0.0
ExplorerHAT       0.4.2
Flask             1.0.2
fourletterphat    0.1.0
gpiozero          1.5.1
html5lib          1.0.1
idna              2.6
isort             4.3.4
itsdangerous      0.24
jedi              0.13.2
Jinja2            2.10
keyring           17.1.1
keyrings.alt      3.1.1
kiwisolver        1.3.1
lazy-object-proxy 1.3.1
logilab-common    1.4.2
lxml              4.3.2
MarkupSafe        1.1.0
matplotlib        3.3.4
mccabe            0.6.1
microdotphat      0.2.1
mote              0.0.4
motephat          0.0.3
mypy              0.670
mypy-extensions   0.4.1
nh-currency       1.0.1
numpy             1.16.2
oauthlib          2.1.0
olefile           0.46
pantilthat        0.0.7
parso             0.3.1
pexpect           4.6.0
pgzero            1.2
phatbeat          0.1.1
pianohat          0.1.0
picamera          1.13
piglow            1.2.5
pigpio            1.44
Pillow            8.1.0
pip               18.1
psutil            5.5.1
pycairo           1.16.2
pycrypto          2.6.1
pycups            1.9.73
pygame            1.9.4.post1
Pygments          2.3.1
PyGObject         3.30.4
pyinotify         0.9.6
PyJWT             1.7.0
pylint            2.2.2
pyOpenSSL         19.0.0
pyparsing         2.4.7
pyserial          3.4
pysmbc            1.0.15.6
python-apt        1.8.4.3
python-dateutil   2.8.1
pyxdg             0.25
PyYAML            5.4.1
rainbowhat        0.1.0
reportlab         3.5.13
requests          2.21.0
requests-oauthlib 1.0.0
responses         0.9.0
roman             2.0.0
**RPi.GPIO          0.7.0    <-------**
RTIMULib          7.2.1
scrollphat        0.0.7
scrollphathd      1.2.1
SecretStorage     2.3.1
Send2Trash        1.5.0
sense-hat         2.2.0
setuptools        40.8.0
simplejson        3.16.0
six               1.12.0
skywriter         0.0.7
sn3218            1.2.7
soupsieve         1.8
spidev            3.4
ssh-import-id     5.7
thonny            3.3.0
touchphat         0.0.1
twython           3.7.0
typed-ast         1.3.1
unicornhathd      0.0.4
urllib3           1.24.1
waveshare-epd     0.0.0
webencodings      0.5.1
Werkzeug          0.14.1
wheel             0.32.3
wrapt             1.10.11

llvllch, 2021-02-28:
Interesting. I'm on 0.6.5. 

Try 
`python3 -m pip install --force-reinstall RPi.GPIO==0.6.5`

It looks like it is a bug in RPi.GPIO in that the line it is complaining about is actually in the script.

manuelautomatisch, 2021-02-28:
> The error pops up when I start the script and cancels the execution of the script.
> 
> Thats the output from **python3 -m pip list**
> 
> Package Version
> 
> asn1crypto 0.24.0
> astroid 2.1.0
> asttokens 1.1.13
> automationhat 0.2.0
> beautifulsoup4 4.7.1
> blinker 1.4
> blinkt 0.1.2
> buttonshim 0.0.2
> Cap1xxx 0.1.3
> certifi 2018.8.24
> chardet 3.0.4
> Click 7.0
> colorama 0.3.7
> colorzero 1.1
> cookies 2.2.1
> cryptography 2.6.1
> cupshelpers 1.0
> cycler 0.10.0
> docutils 0.14
> drumhat 0.1.0
> entrypoints 0.3
> envirophat 1.0.0
> ExplorerHAT 0.4.2
> Flask 1.0.2
> fourletterphat 0.1.0
> gpiozero 1.5.1
> html5lib 1.0.1
> idna 2.6
> isort 4.3.4
> itsdangerous 0.24
> jedi 0.13.2
> Jinja2 2.10
> keyring 17.1.1
> keyrings.alt 3.1.1
> kiwisolver 1.3.1
> lazy-object-proxy 1.3.1
> logilab-common 1.4.2
> lxml 4.3.2
> MarkupSafe 1.1.0
> matplotlib 3.3.4
> mccabe 0.6.1
> microdotphat 0.2.1
> mote 0.0.4
> motephat 0.0.3
> mypy 0.670
> mypy-extensions 0.4.1
> nh-currency 1.0.1
> numpy 1.16.2
> oauthlib 2.1.0
> olefile 0.46
> pantilthat 0.0.7
> parso 0.3.1
> pexpect 4.6.0
> pgzero 1.2
> phatbeat 0.1.1
> pianohat 0.1.0
> picamera 1.13
> piglow 1.2.5
> pigpio 1.44
> Pillow 8.1.0
> pip 18.1
> psutil 5.5.1
> pycairo 1.16.2
> pycrypto 2.6.1
> pycups 1.9.73
> pygame 1.9.4.post1
> Pygments 2.3.1
> PyGObject 3.30.4
> pyinotify 0.9.6
> PyJWT 1.7.0
> pylint 2.2.2
> pyOpenSSL 19.0.0
> pyparsing 2.4.7
> pyserial 3.4
> pysmbc 1.0.15.6
> python-apt 1.8.4.3
> python-dateutil 2.8.1
> pyxdg 0.25
> PyYAML 5.4.1
> rainbowhat 0.1.0
> reportlab 3.5.13
> requests 2.21.0
> requests-oauthlib 1.0.0
> responses 0.9.0
> roman 2.0.0
> **RPi.GPIO 0.7.0 <-------**
> RTIMULib 7.2.1
> scrollphat 0.0.7
> scrollphathd 1.2.1
> SecretStorage 2.3.1
> Send2Trash 1.5.0
> sense-hat 2.2.0
> setuptools 40.8.0
> simplejson 3.16.0
> six 1.12.0
> skywriter 0.0.7
> sn3218 1.2.7
> soupsieve 1.8
> spidev 3.4
> ssh-import-id 5.7
> thonny 3.3.0
> touchphat 0.0.1
> twython 3.7.0
> typed-ast 1.3.1
> unicornhathd 0.0.4
> urllib3 1.24.1
> waveshare-epd 0.0.0
> webencodings 0.5.1
> Werkzeug 0.14.1
> wheel 0.32.3
> wrapt 1.10.11

llvllch, 2021-02-28:
I have RPi.GPIO version 0.6.5..... try downgrading to that

manuelautomatisch, 2021-02-28:
Okay, I tried it but the same error pops up again

llvllch, 2021-02-28:
I added a cleanup line to the exit of the code, perhaps that helps? Try it on the latest version.

manuelautomatisch, 2021-02-28:
Okay, I'll try

manuelautomatisch, 2021-02-28:
No seems not to work, same error again.

DEBUG:root:spi end
DEBUG:root:close 5V, Module enters 0 power consumption ...
Traceback (most recent call last):
  File "btcticker.py", line 364, in <module>
    main()
  File "btcticker.py", line 317, in main
    key1state = GPIO.input(key1)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)

llvllch, 2021-02-28:
Given that I can't reproduce, and that script does set pin numbering mode (line 298), I'd either raise it with a RPi.GPIO forum and/or do a clean install.

Unless anyone else can see the issue?

manuelautomatisch, 2021-02-28:
Okay, thank you anyway. I will see what im doing

llvllch, 2021-02-28:
I'll leave it open in the hope that someone else has the answer :)

manuelautomatisch, 2021-02-28:
okay, thx :)

llvllch, 2021-02-28:
Found it! 

Latest pull will fix it

dkim777, 2021-02-28:
Latest code works for me, thanks!
by the way my RPi.GPIO is also 0.7.0

closed 2021-03-01 by llvllch

manuelautomatisch, 2021-03-01:
Ah nice, latest code works for me too. Thank you very much :)