veebch/btcticker #38: Crash on run

15rhema, 2021-03-28:
Hello,

I'm using a PI zero and I can't get the application to run using the HAT GPIO connector. Connected via SSH I run the application and see this: 
[image]

Everything runs normal when I use the 8 pin connector. Running the latest release. I've tried with 2 different PI Zeros, same result.

llvllch, 2021-03-28:
Hey! I've never seen this before. Possible hardware error on the screen?

15rhema, 2021-03-28:
Forgot to mention this is the second screen I have have tried too. Same issue. Maybe I just have really bad luck with the screens?

This is the screen I bought:

https://www.amazon.com/2-7inch-HAT-Resolution-Electronic-Communicating/dp/B075FQKSZ9/ref=sr_1_12?dchild=1&keywords=waveshare+e-ink&qid=1616958141&sr=8-12

and I'm using this power supply:

https://www.amazon.com/gp/product/B083JCL6H2/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

llvllch, 2021-03-28:
That's the screen I've used with a pi zero. Have you tried a fresh install of the Waveshare Module?

closed 2021-04-06 by llvllch