veebch/btcticker #41: python script crashes

profwww, 2021-04-13:
First I would like to thank you for your awesome work in making and sharing the btc ticker.

Now regarding the issue.

A few Hours after running `python3 btcticker.py` flawlessly inside a screen session it crashes and here is the traceback of the error that it gives me, taken from the command line.

```
    pricestack, ATH = getData(config, other)
  File "btcticker/btcticker.py", line 110, in getData
    rawtimeseries = requests.get(geckourlhistorical).json()
  File "/usr/lib/python3/dist-packages/requests/models.py", line 897, in json
    return complexjson.loads(self.text, **kwargs)
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "btcticker/btcticker.py", line 364, in <module>
    main()
  File "btcticker/btcticker.py", line 349, in main
    lastcoinfetch=fullupdate(epd,config)
  File "btcticker/btcticker.py", line 282, in fullupdate
    lastgrab=lastcoinfetch
NameError: name 'lastcoinfetch' is not defined
pi@pizero-ticker:~ $ 
```

llvllch, 2021-04-14:
Too kind! 

That error is triggered by the coingecko api not returning data for some reason. Should be a quick fix. I'll leave this open until I have.

profwww, 2021-04-14:
Thanks, I'll keep running the latest version for two more days and if there's no crash I will close this issue.

llvllch, 2021-04-14:
Great, the test will be when it gets no data and the 'bean screen' shows for 20 seconds before it tries again. I've seen it do it once today so far.

profwww, 2021-04-15:
Yes same with me, A few seconds after the bean screen it tries again and works. Issue fixed.

closed 2021-04-15 by profwww