veebch/btcticker #42: python3 -m pip install -r requirements.txt

flyingmonkey12, 2021-04-15:
Hello all, I am running a fresh install of Rasbian on Pi Zero W and using waveshare 2.7in e-paper HAT. Whenever I run "python3 -m pip install -r requirements.txt", it throw me this error:
 
`  Failed building wheel for numpy `
`  Running setup.py clean for numpy `
`  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokeni        ze;__file__='/tmp/pip-install-hecy4lsy/numpy/setup.py';f=getattr(tokenize, 'open        ', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(co        de, __file__, 'exec'))" clean --all:`
`  Running from numpy source directory.`
`
  `setup.py clean` is not supported, use one of the following instead:`

    - `git clean -xdf` (cleans all files)
    - `git clean -Xdf` (cleans all versioned files, doesn't touch
                        files that aren't checked into the git repo)
`
`  Add `--force` to your command to use it anyway if you must (unsupported).`
`

1. I have tried using full Rasbian
2. I have followed and installed everything I could google.
3. I started over so many times
4. pi@raspberrypi:~ $ pip3 list
Package         Version
--------------- ---------
asn1crypto      0.24.0
certifi         2018.8.24
chardet         3.0.4
cryptography    2.6.1
cycler          0.10.0
Cython          0.29.23
entrypoints     0.3
idna            2.6
Jetson.GPIO     2.0.16
keyring         17.1.1
keyrings.alt    3.1.1
kiwisolver      1.3.1
matplotlib      3.4.1
nh-currency     1.0.1
numpy           1.20.2
olefile         0.46
Pillow          8.2.0
pip             18.1
pycrypto        2.6.1
PyGObject       3.30.4
pyparsing       2.4.7
python-apt      1.8.4.3
python-dateutil 2.8.1
pyxdg           0.25
PyYAML          5.4.1
requests        2.12.4
RPi.GPIO        0.7.0
SecretStorage   2.3.1
setuptools      40.8.0
six             1.12.0
spidev          3.5
ssh-import-id   5.7
urllib3         1.24.1
waveshare-epd   0.0.0
wheel           0.32.3

5. when I run python3 btcticker.py, I get following
pi@raspberrypi:~/btcticker $ python3 btcticker.py
Traceback (most recent call last):
  File "btcticker.py", line 9, in <module>
    import RPi.GPIO as GPIO
  File "/usr/local/lib/python3.7/dist-packages/RPi/GPIO/__init__.py", line 1, in <module>
    from Jetson.GPIO import *
  File "/usr/local/lib/python3.7/dist-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/usr/local/lib/python3.7/dist-packages/Jetson/GPIO/gpio.py", line 70, in <module>
    model, JETSON_INFO, _channel_data_by_mode = gpio_pin_data.get_data()
  File "/usr/local/lib/python3.7/dist-packages/Jetson/GPIO/gpio_pin_data.py", line 438, in get_data
    raise Exception('Could not determine Jetson model')
Exception: Could not determine Jetson model

6. I also read this link: <https://github.com/llvllch/btcticker/issues/13>

Can someone help me troubleshoot this?  What am I missing?

llvllch, 2021-04-15:
That looks like you've installed the Jetson version of the GPIO module (See the messages in #5 on your list). If you revisit the waveshare install, you might have some joy

flyingmonkey12, 2021-04-15:
1. so I went to waveshare's github from their Demo code: https://github.com/waveshare/e-Paper
2. then, I downloaded their "RaspberryPi_JetsonNano"
3. I used their instruction from their pdf: <https://www.waveshare.com/w/upload/3/31/2.7inch_e-paper_hat_user_manual_en.pdf>
4. I did everything from page 13 to page 15.  But, I didn't run anything listed below WIRINGPI LIBRARY and PYTHON2 LIBRARY. I assumed I just need PYTHON3... so I ran everything under that...
6. I ran sudo pip3 install RPi.GPIO, it says
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: RPi.GPIO in /usr/local/lib/python3.7/dist-packages (0.7.0)

I know I must done something wrong, but I have no clue what is it that I am missing...? Sorry I am 2 days old with python experience...

llvllch, 2021-04-15:
>    5. I ran sudo pip3 install RPi.GPIO, it says

looks fine! It is just saying you already have RPi.GPIO.... keep going :)

flyingmonkey12, 2021-04-15:
pi@raspberrypi:~/btcticker $ python3 btcticker.py
Traceback (most recent call last):
  File "btcticker.py", line 9, in <module>
    import RPi.GPIO as GPIO
  File "/usr/local/lib/python3.7/dist-packages/RPi/GPIO/__init__.py", line 1, in <module>
    from Jetson.GPIO import *
  File "/usr/local/lib/python3.7/dist-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/usr/local/lib/python3.7/dist-packages/Jetson/GPIO/gpio.py", line 70, in <module>
    model, JETSON_INFO, _channel_data_by_mode = gpio_pin_data.get_data()
  File "/usr/local/lib/python3.7/dist-packages/Jetson/GPIO/gpio_pin_data.py", line 438, in get_data
    raise Exception('Could not determine Jetson model')
Exception: Could not determine Jetson model

llvllch, 2021-04-15:
looks like the jetson version is still there... could you restart from a
clean image?

flyingmonkey12, 2021-04-15:
not sure if I need to be concerned...
so I ran from fresh image, while sudo tar zxvf bcm2835-1.xx.tar.gz, then extract, then configure, then "make", I get the following:
pi@raspberrypi:~/pi/bcm2835-1.68 $ make
make  all-recursive
make[1]: Entering directory '/home/pi/pi/bcm2835-1.68'
Making all in src
make[2]: Entering directory '/home/pi/pi/bcm2835-1.68/src'
gcc -DHAVE_CONFIG_H -I. -I..     -g -O2 -MT bcm2835.o -MD -MP -MF .deps/bcm2835.Tpo -c -o bcm2835.o bcm2835.c
bcm2835.c:1945:1: fatal error: opening dependency file .deps/bcm2835.Tpo: Permission denied
 }
 ^
compilation terminated.
make[2]: *** [Makefile:567: bcm2835.o] Error 1
make[2]: Leaving directory '/home/pi/pi/bcm2835-1.68/src'
make[1]: *** [Makefile:365: all-recursive] Error 1
make[1]: Leaving directory '/home/pi/pi/bcm2835-1.68'
make: *** [Makefile:306: all] Error 2

llvllch, 2021-04-16:
The waveshare instructions tell you to run 'sudo make' :)

flyingmonkey12, 2021-04-16:
(ಥ﹏ಥ)  I promise you the pdf didn't say "sudo make" on page 14: https://www.waveshare.com/w/upload/3/31/2.7inch_e-paper_hat_user_manual_en.pdf

so I started from scratch clean Pi lite and I did the following:
sudo apt-get update
sudo apt-get upgrade
sudo apt install git
sudo apt install screen

sudo cp -r /boot/RaspberryPi/ ./
sudo chmod 777 -R RaspberryPi/
sudo tar zxvf bcm2835-1.68.tar.gz
cd bcm2835-1.68
sudo ./configure
sudo make
sudo make check
sudo make install

sudo apt-get install python3-pip 
sudo apt-get install python3-pil
sudo pip3 install spidev
sudo pip3 install RPi.GPIO
sudo pip3 install Pillow
sudo pip3 install nh-currency
sudo apt-get install python3-numpy

sudo apt-get install libopenjp2-7
sudo apt-get install libqt5gui5
sudo apt-get install python-scipy
sudo apt install libatlas-base-dev

cd e-Paper/RaspberryPi_JetsonNano/python
sudo python3 setup.py install
cd ~
git clone https://github.com/llvllch/btcticker.git
cd btcticker
python3 -m pip install -r requirements.txt
screen bash
python3 btcticker.py

pi@raspberrypi:~/btcticker $ python3 btcticker.py
Traceback (most recent call last):
  File "btcticker.py", line 364, in <module>
    main()
  File "btcticker.py", line 296, in main
    epd.Init_4Gray()
  File "/usr/local/lib/python3.7/dist-packages/waveshare_epd-0.0.0-py3.7.egg/waveshare_epd/epd2in7.py", line 295, in Init_4Gray
    if (epdconfig.module_init() != 0):
  File "/usr/local/lib/python3.7/dist-packages/waveshare_epd-0.0.0-py3.7.egg/waveshare_epd/epdconfig.py", line 74, in module_init
    self.SPI.open(0, 0)
FileNotFoundError: [Errno 2] No such file or directory

llvllch, 2021-04-16:
Did you enable SPI?

flyingmonkey12, 2021-04-16:
(づ｡◕‿‿◕｡)づ omg!!  this is my first time seeing the screen actually works!!!  I am now too scared to power it off, it is looking so pretty.
Thank you so much, I will give you a hug if I physically can!!
One more question: does this ticker just automatically run the "python3 btcticker.py" on power up? 
I am building this little thing for my grandpa, he has invested into some btc. I doubt he knows how to ssh.

llvllch, 2021-04-16:
Hooray! To make it run on power up, I used systemctrl. You can also use
cronjobs…..

flyingmonkey12, 2021-04-16:
I have unplugged it from outlet power and had it plugged to my laptop USB. It is alive and working! 
sudo nano /etc/rc.local
add below line above exit0
sudo -u pi /usr/bin/python3 /home/pi/btcticker/btcticker.py &
ctrl + x
Y
Enter
it now auto-magically runs when powering on.  Thanks for all your help!!

closed 2021-04-16 by flyingmonkey12

llvllch, 2021-04-17:
Bravo! Your grandad will be pleased!