veebch/btcticker #43: Numpy issues after running "python3 -m pip install -r requirements.txt"

JosephMacReady, 2021-04-15:
Hi, and thanks **a lot** for sharing your work on this!! I'm very excited to try and get it running on my Pi.

The problems I have start after running the `python3 -m pip install -r requirements.txt` even though I did run the four troubleshoot lines successfully.

The process starts but then gets to 
```
Building wheels for collected packages: numpy
Running setup.py bdist_wheel for numpy 
```
and stays on that for a long period of time, after which it fails. Here is the output, I cut out two long parts to point out what seems the most important.

Please let me know if you have encountered this and know what the culprit may be!
Joe M.

Running setup.py bdist_wheel for numpy ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-xg_zeivx/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-ye9o1mce --python-tag cp37:
  Running from numpy source directory.

[...]

  ----------------------------------------
  **Failed building wheel for numpy
  Running setup.py clean for numpy**
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-xg_zeivx/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" clean --all:
  Running from numpy source directory.
  
  `setup.py clean` is not supported, use one of the following instead:
  
    - `git clean -xdf` (cleans all files)
    - `git clean -Xdf` (cleans all versioned files, doesn't touch
                        files that aren't checked into the git repo)
  
  Add `--force` to your command to use it anyway if you must (unsupported).
  
  ----------------------------------------
  **Failed cleaning build dir for numpy
Failed to build numpy**
Installing collected packages: numpy, kiwisolver, python-dateutil, cycler, pyparsing, matplotlib, Pillow, PyYAML, RPi, nh-currency
  Running setup.py install for numpy ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-xg_zeivx/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-zxsxf_7k/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    Running from numpy source directory.

[...]

    ----------------------------------------
**Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-xg_zeivx/numpy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-zxsxf_7k/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-install-xg_zeivx/numpy/**

llvllch, 2021-04-15:
Pleasure! I'm pretty sure the code used to generate requirements.txt is misbehaving. An easy way around is to try installing the modules listed in the requirements.txt file one by one (eg 'python3 -m pip install Pillow' etc). In the meantime, I'll try to fix it.

JosephMacReady, 2021-04-15:
> Pleasure! I'm pretty sure the code used to generate requirements.txt is misbehaving. An easy way around is to try installing the modules listed in the requirements.txt file one by one (eg 'python3 -m pip install Pillow' etc). In the meantime, I'll try to fix it.

Thank you. I will try to do so now and report back.

llvllch, 2021-04-15:
OK, I also updated requirements.txt, that might help

JosephMacReady, 2021-04-15:
Great, and the separate installs worked on my end; so all squared away, thank you

closed 2021-04-15 by JosephMacReady