veebch/btcticker #47: IDEA: Speed up the display.

RattyDAVE, 2021-04-26:
the writing to the display is a bit slow. (Well way more than a bit)

This module https://github.com/elad661/rpi_epd2in7 has a much quicker write method.

llvllch, 2021-04-26:
That's a good idea. 

I vaguely recall that tried this one, thinking the partial refresh would a better way to do it. In the end I opted for the 'official' waveshare module due to the lack of activity on that module.

closed 2021-06-12 by llvllch

NeilInnes, 2021-06-15:
Thoughts on this module @llvllch https://github.com/MarvelousBlack/rpi_epaper ?

Happy to give it a whirl if this module seem ok to you 😄

llvllch, 2021-06-15:
Yes please. I'm going to add a bit of backoff retry logic to the data pull.
Coingecko seems to throttle during times of high traffic, so I'll try to
make the data pull a little more mindful of that

NeilInnes, 2021-06-15:
Cool - got it working last night to use this library - partial update for this screen (according to the library) only supports a max area of 255x255 so will need quite a refactor to do this. Leave it with me 😄

RattyDAVE, 2021-06-15:
I have been looking at this. https://github.com/pskowronek/epaper-clock-and-more/tree/master/epds

This is changes the timings so quicker. About 10x quicker on Black and white screen.

NeilInnes, 2021-06-15:
I like the backoff idea - this could be:
- Move all the API calls are moved to a central new method
- The method could try the request and on fail (HTTP code, timeout or message):
  - sleep (attempt count * 5 seconds?)
  - Retry (recursively call itself)
  - Increase a retry count variable (passed recursively + 1)
  - On reaching maximum number of retries, return an error (as current functionality)

NeilInnes, 2021-06-15:
> I have been looking at this. https://github.com/pskowronek/epaper-clock-and-more/tree/master/epds
> 
> This is changes the timings so quicker. About 10x quicker on Black and white screen.

That appears to use FAST mode - from other projects I've read `using fast mode may make your e-paper display permanently unrecoverable damage` 😟

RattyDAVE, 2021-06-15:
Ahhh.. maybe not then!

NeilInnes, 2021-06-15:
For a device that isn't on continuously it may be ok - but I'd prefer it to update slower than get fried 😄

llvllch, 2021-06-15:
This was my issue, on a cycling ticker (more than one coin) the date changes, the token image, the price, and albeit a tiny bit, the sparkline. Throw in the fact that partial refresh often leads to ghosting and needs to fully refreshed every now and again made me think "I'll use the manufacturers code"

E-paper is slow. but on a refresh interval of every 5-10 mins, I barely notice it (unless it's in my line of sight at the time)

llvllch, 2021-06-15:
> I like the backoff idea - this could be:
>
>     * Move all the API calls are moved to a central new method
>
>     * The method could try the request and on fail (HTTP code, timeout or [...]

Pretty much, although it doubles sleep time each count. I've committed my first stab at it.

NeilInnes, 2021-06-15:
I've got a prototype of this working - I split the image into 2 parts and updated them sequentially. I've tried both alternative edp2in7 libraries too.

However, partial updates only support black+white - not 4 gray mode. This means the logo and spark line end up ghosting over each other (not so much of an issue if you aren't cycling through different coins of course). 😞

llvllch, 2021-06-16:
e-paper gonna e-paper :)

NeilInnes, 2021-06-16:
lol - think I'll spend a bit more time looking around, but if there if nothing comes up this may need to be closed under "not feasible"

NeilInnes, 2021-07-08:
I'm calling this one - I've tried around 6 different libraries for the display. All have issues with partial updates and even the best ones leaving too much ghosting when cycling coins. The only solution here is a different type of display - but then, it would lose it's charm 🥰

llvllch, 2021-07-08:
A noble effort! My simplistic brain tells me that no one loves Waveshare
screens as much as Waveshare does, so their code is probably aimed at
preserving the hardware.