veebch/btcticker #49: Some tokens not working

pep987, 2021-05-01:
I have an issue with Polkadot, Ocean-Token and Kusama tokens which for some reason do not show, I get the image of the bean instead. Bitcoin, Ethereum and others work fine.

llvllch, 2021-05-01:
Are you using the names as listed on here:
https://www.coingecko.com/api/documentations/v3#/coins/get_coins_list
?

pep987, 2021-05-01:
Does it use the ID or name? I have tried for instance Polkadot and polkadot and neither appear to work for me.

{
    "id": "polkadot",
    "symbol": "dot",
    "name": "Polkadot"
  }

I am using usd,gbp,eur for fiatcurrency, not sure if that makes any difference.

llvllch, 2021-05-01:
is the exchange 'default'?

pep987, 2021-05-01:
Yes.

llvllch, 2021-05-01:
I just ran it with ‘polkadot’, it worked. Are you running the latest
version?

pep987, 2021-05-01:
I thought I was but I need to check to be honest, I only installed it Thursday though so I assume I am, if its working for you though it must be me missing something somewhere.

llvllch, 2021-05-01:
could it be a memory issue? If the script hasnt seen the coin before, it pulls and saves the image

pep987, 2021-05-01:
Using a pi zero with a 32GB SD card. It does appear to run the code correctly if I watch it, but shows the bean instead of the ticker, I have energy web token working okay and that likely had to pull the image. Most strange.

llvllch, 2021-05-01:
can you paste a copy of your config.yaml file?

pep987, 2021-05-01:
The actual file or contents of it?

llvllch, 2021-05-01:
just the contents, also, does the beanscreen have an error message
underneath and the time at the top?

pep987, 2021-05-01:
display:
  cycle: true
  inverted: false
  orientation: 270
ticker:
  currency: theta-token,polkadot,energy-web-token,ecomi,bitcoin,ethereum,polkadot,kusama
  exchange: default
  fiatcurrency: usd,gbp,eur
  sparklinedays: 14
  updatefrequency: 180

The bean picture has no time at the top, or any error message, just a picture of a bean scooting across the screen with a bit of a shadow below the image. The tickers shows the ones that are working, and when it gets to the ones it does not like, it just shows the bean image and then moves on to the next one.

pep987, 2021-05-01:
[image]

pep987, 2021-05-01:
Image of the bean

llvllch, 2021-05-01:
Mystery solved.....thats an older version of the code

pep987, 2021-05-01:
Ahh in that case I apologise, I am indeed an idiot.
I installed it via the info on the main page including git clone https://github.com/llvllch/btcticker.git command, how do I get the latest version?
Also can I ask is there an easy way to make the symbol smaller and the chart and info larger?

pep987, 2021-05-01:
Also just to confirm, I am using the btcticker2in13.py code as I have the smaller screen not the 2.7"

llvllch, 2021-05-01:
Ah! thats the issue! The code on that needs updating

pep987, 2021-05-03:
That explains it. Do you think you will convert the code for the 2in13 at any point?

I had a go but got stuck.

llvllch, 2021-05-03:
Sure, it shoulnt be hard, as soon as I get a moment. That code was
contributed so I’ve not fiddled with it yet. All you would need to do is
add the couple of lines that pull the image if it is not already there - an
even easier way would be to make your own images for those coins and put
them in the directory for the tokens. No coding required!

pep987, 2021-05-03:
I got it mostly working but was not sure how to define this:

def display_image(epd, img):
    epd.display_4Gray(epd.getbuffer_4Gray(img))

I think I might have added to much from what you have put there, I do now get the bean with the error though, although the error shows reversed

pep987, 2021-05-03:
I modified the file, its now working for all tokens.
Thank you for your prompt replies.

closed 2021-05-03 by pep987

llvllch, 2021-05-03:
Great!

Want to add your changes to the repository? :)

pep987, 2021-05-03:
Sure, will do. Sorry, should of thought of that.

pep987, 2021-05-05:
I have an interesting update. I still see the bean image, however it is loading all the tokens, I have checked the logging and I can see it load a certain token, it displays it and all is good, then it shows the bean, then loads the next token and displays that, and there are no errors in between the two. Will have to do some more looking into why it does that, like I said it loads all tokens fine, and downloads all the correct token images without issue.