veebch/btcticker #50: When in either landscape orientation, imaged is mirrored (i.e. reads backwards)

psychosomatico, 2021-05-13:
Hi,

Sorry, I have almost no programming experience and first time using a Pi. I just wanted to try my hand with this cool project. Everything seems to be working except that the image is completely reversed (not inverted like what the 3rd button does). It is like the image is its mirror image. However, this is only in either of the two landscape rotations. When I rotate 90 degrees into portrait, or 180 to portrait the other way, its normal.

Thanks!

llvllch, 2021-05-13:
Interesting! If you comment out line 236 this will prevent the mirroring.
ie change

image = ImageOps.mirror(image)

to

# image = ImageOps.mirror(image)
in the file btcticker.py

psychosomatico, 2021-05-13:
Thank you! You rock!

closed 2021-05-13 by psychosomatico

lborruto, 2021-05-26:
I tried this but getting this error after commenting the line : 

`sudo python3 btcticker.py
  File "btcticker.py", line 242
    tokenimage.close()
             ^
IndentationError: expected an indented block`

llvllch, 2021-05-26:
Ahhh, that's a python thing. Make sure it's indented correctly.

lborruto, 2021-05-26:
> ,

Oh ok, didn't know it was so sensitive ! Thanks for the fast answer :)