veebch/btcticker #52: Script / Display freezes after certain time

MPower-max, 2021-06-10:
After following each step from this repository (incl. installation of additional libs from inititial trouble shooting), I was able to get the btcticker running (i used the same hardware, but instead of remote ssh access, I went for a 32 bit Raspian image and installed everything directly on the desktop with the console).
  
Great work and fantastic features! It is exactly the device I was looking for with only a few minor flaws:

I use the cycling option with 5 different crypto coins. After a certain time (it seems to be reproducable), the display gets stuck. After a restart (unplugging and plugging in the power cable), everything works normal again.

I wrote a small script and added it to crontab on the raspi, so that it starts the btcticker automatically on start up and it also restarts the Rapi slightly before the program usually crashes. This works absolutely fine for me at the moment, but I thought there might be an easier (maybe less stressful for the Raspi) solution for the issue with the program getting stuck?

Moreover, it would be great to have a simple way to shut down the Raspi properly (instead of only unplugging the power USB cable). Do you think it would be possible to include that option in combination with e.g. a button press-and-hold on the e-ink display?

Many thanks (please excuse if I have missed an easy solution as I am quite new to programming)!

Cheers, Max

llvllch, 2021-06-10:
This is excellent to hear! Good to see that you're coding...

I've not been able to get that error. Do you get an error message when it happens? I think that for faster refresh rates, coingecko may stop returning data. 

One of the reasons I might not be seeing it is that I used systemd for process control, and I think it may be handling the restarts for me.

The power button is a good idea. I'll have a look at adding something. Let me know if you come up with anything good :)

closed 2021-06-12 by llvllch

MPower-max, 2021-06-12:
Sorry for not responding earlier ...

Actually, I tried to reproduce the error in the last few days and it seems that it only occurs under the following conditions:

- script is started automatically via crontab
- no inputs (e.g. via keyboard) while the device is running
- no other errors occur (such as a lost connection to the Wifi or coingecko not available)

Maybe the Raspian has a hidden sleep or hibernate mode when there are absolutely no inputs after start up, because when the display freezes the Raspi status LED blinks very consistent (same time on and off), but I could not find any additional information for my assumption.

I attached also two images from errors I saw recently:
First photo (after freezing) was displayed, as the title indicates, after I left it frozen for some time (if I remember correctly it worked again later on without any tempering).

Second photo was taken later on that evening (I guess it was only because it lost connection somehow).

Cheers, Max 

[image]

[image]

llvllch, 2021-06-12:
The first error is an ugly handling of coingecko not returning data and the script trying to draw a graph of no data. I suspect the second is also a coingecko issue. 

Bonus: if you pull that little green tab to the left of the screen, the shiny screen protector comes off. It will be much prettier then :)

MPower-max, 2021-06-12:
Many thanks for the prompt explanation and the tip with screen protector 👍 ! 
As long as I need to solder on the back of the Raspi (with the screen face down on my work bench) I prefer to let it stay in place ;) (I added a small battery (21700) and battery management to run the btcticker for about a day without power supply) and I have not found a proper casing for all the parts.

Cheers

MPower-max, 2021-06-13:
btw ... as I mentioned earlier, I highly appreciate the work you put into the BTC Ticker. Is it possible to donate some coins (ETH, DOGE, BTC Cash, XRP, ... ) to support your work?