veebch/btcticker #57: Stuck on INFO:root:Getting token Image from Image directory

reddai75, 2021-06-11:
Hello!
that's a great project!
I'm trying to make the project work with a RPI zero W and I followed your instructions. I get stuck to 
INFO:root:Getting token Image from Image directory
and after a few minutes my ssh connection drops. I don't think is the power adapter since it powers a RPI4 flawlessy....
could you help me ?
many thanks!

reddai75, 2021-06-11:
Hello! Thanks for the quick answer! 
I used Bitcoin... I think I made a mistake somewhere during the installation process. I followed your instructions, first from here https://www.waveshare.com/wiki/2.7inch_e-Paper_HAT
I enabled SPI
then Installed BCM2835 libraries, wiringPi libraries,Install Python libraries, git, I also downloaded the examples.
then as you wrote:
cd e-Paper/RaspberryPi_JetsonNano/python
sudo python3 setup.py install
then I followed the remaining instructions ( I did not install screen, is this the mistake ?)
I'm on a Raspberry PI zero W headless with a 2.7 inch e-paper hat from waveshare.

it hangs too also with the examples :(
thanks again!
Marco

llvllch, 2021-06-11:
Hey Marco. It sounds like you might have missed a step on the waveshare stuff, if their libraries aren't working either. 

The hardware you describe is the configuration I use and works without problems. Was it the light version of Rasbian you used? Could it be a missing library?

reddai75, 2021-06-11:
yes I used the light version…
so you have any suggestions?

llvllch, 2021-06-11:
Perhaps try the advice under Troubleshooting on the Readme file ie:

```
sudo apt-get install libopenjp2-7
sudo apt-get install libqt5gui5
sudo apt-get install python-scipy
sudo apt install libatlas-base-dev
```

reddai75, 2021-06-11:
I did it when I installed it 😩

llvllch, 2021-06-11:
I'd be tempted to start again. I used that exact setup to build an image from scratch a few days ago. It could be a hardware problem, but that's less likely than just a simple mistake.

One thing about the Python libraries step on the waveshare wiki: I only install the Python 3 part....

reddai75, 2021-06-11:
ok ! will do it ! thanks

llvllch, 2021-06-11:
OK.... I'll leave this open until you get it working :)

DickWhitmanBTC, 2021-06-11:
Do you happen to have one that works on a larger epaper display?

reddai75, 2021-06-11:
It seems to work now. The problem is probably with the GPIO female not correctly soldered. I connected the display to the raspberry with the cables and it works... BUT, the buttons do not :(

llvllch, 2021-06-12:
Well that's something. If you can't use the buttons, you can always enable cycle mode so that it automatically scrolls through coins. The orientation and inversion can both also be set in the config file. So at present, the only thing you uses is cycling through currencies. 

Anyway, glad you made progress! Closing this issue as it is pretty certainly a waveshare thing.

closed 2021-06-12 by llvllch