veebch/btcticker #62: 2.7inch e-Paper e-Ink Display HAT (B) For Raspberry Pi, 264×176, "Red / Black / White, SPIÉis that there is the possibility to make it work with 2.7inch e-Paper e-Ink Display HAT (B) For Raspberry Pi, 264×176, Red / Black / White, SPI"

dish67, 2021-06-23:
Hi

Is that there is the possibility to make it work with 2.7inch e-Paper e-Ink Display HAT (B) For Raspberry Pi, 264×176, Red / Black / White, SPI

I change parameter  on btcticker.py

from waveshare_epd import epd2in7
**to** 
from waveshare_epd import epd2in7b_V2

**and** 
    epd = epd2in7b_V2.EPD()
#epd = epd2in7()
    epd.init()
#    epd.Init_4Gray()
#    epd.display_4Gray(epd.getbuffer_4Gray(img))
#    epd.display(epd.getbuffer(img))

But I did't see any image on my e-paper ... 

do you have any suggestion ??

llvllch, 2021-06-23:
Uncomment the line: 
#epd.display(epd.getbuffer(img))
?

dish67, 2021-06-23:
Thank .. 
But I have this error !!  TypeError: display() missing 1 required positional argument: 'imagered'   <==

when I Uncomment the line and reexecute  
def display_image(img):
    epd = epd2in7b_V2.EPD()    epd.init()    print("Clear...")#    epd.Init_4Gray()#    epd.display_4Gray(epd.getbuffer_4Gray(img))    epd.display(epd.getbuffer(img))    epd.sleep()    initkeys()    return

python3 btcticker.py ...Clear...DEBUG:root:HorizontalTraceback (most recent call last):  File "btcticker.py", line 360, in fullupdate    display_image(image)  File "btcticker.py", line 319, in display_image    epd.display(epd.getbuffer(img))TypeError: display() missing 1 required positional argument: 'imagered'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):  File "btcticker.py", line 454, in main    lastcoinfetch=fullupdate(config,lastcoinfetch)  File "btcticker.py", line 366, in fullupdate    display_image(image)  File "btcticker.py", line 319, in display_image    epd.display(epd.getbuffer(img))TypeError: display() missing 1 required positional argument: 'imagered'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):  File "btcticker.py", line 473, in <module>    main()  File "btcticker.py", line 463, in main    display_image(image)  File "btcticker.py", line 319, in display_image    epd.display(epd.getbuffer(img))TypeError: display() missing 1 required positional argument: 'imagered'   <==

Michel Thibault

***@***.***

385 de gascogne ,

J4B 8T2 , Boucherville 

Quebec , Canada

llvllch, 2021-06-23:
This is where it looks like it's different 'image red' sounds like you might have a separate image that determines the red part of the image

closed 2021-07-21 by llvllch