veebch/btcticker #64: After 24/36 hrs service stops

reddai75, 2021-06-26:
Hello, 
I have updated btcticker to the latest version, but I still experience a crash (not sure if "crash" is the correct word) every now and then.
I am not a programmer but if there is a way I can contribute in finding the issue I would be happy to help...

llvllch, 2021-06-26:
By crash, do you mean freeze, or does an error screen show? I made a small
but significant change this morning, try updating to that (and thank you
for the help!)

reddai75, 2021-06-26:
it depends: for example this morning the pi zero was so slow that I couldn't log in: I inserted username , then after I inserted the password the connection timed out... a couple of days ago I was able to log in but the service was probably stopped or frozen: I checked the status (service btcticker status) and it displayed the last update that occured a couple of hours earlier.
I'll update it over the weekend and let you know! 
Thank you for this project!

llvllch, 2021-06-26:
Try the new version. Someone kindly pointed out that the main loop was
generating a lot of network traffic. That's fixed now (I think)

reddai75, 2021-06-29:
Hello, I have tried the new version since Sunday, and since then two crashes occured ... I have put btcticker as a service at startup, and here is what I have just get this morning after service btcticker status... as you can see last update occured  around midnight... then nothing.
[image]

llvllch, 2021-06-29:
Check dmesg and /var/log/syslog to see if there are any clues as to what
happened.

reddai75, 2021-07-05:
quick update: now it runs for a couple of days without problems, but after that every now and then the pi zero led starts blinking and there is no screen refresh for up to even 10 hours or more. Later in the day everything goes back to normal… that’s weird.

llvllch, 2021-07-06:
That is weird, I've seen the flashes too, but google has nothing about
them. I suspect it could be a power saving mode. I'll leave this open and
keep investigating.

closed 2021-07-21 by llvllch