veebch/btcticker #65: Start at bootup?

docjfnoel, 2021-06-28:
Hello,
Sorry if it's been answered or written anywhere but is there any script somewhere to start the ticker at bootup?  
Thanks!

llvllch, 2021-06-28:
The easiest thing would be to set a cronjob to run at boot. Slightly more
sophisticated would be to let systemd take care of things. Both are
described here:
https://www.cyberciti.biz/faq/linux-execute-cron-job-after-system-reboot/

docjfnoel, 2021-06-30:
Thanks for your help!
I must be dumb beginner but I've had no luck with a couple of methods to start python script on boot...
Could you point a specific code?

llvllch, 2021-07-01:
We're all dumb beginners!

Sure, the most simple would be to type
       crontab -e
from a prompt
       and then in the file add the line
      @reboot /usr/local/bin/python3 /home/pi/btcticker/btcticker.py
then save the file, done!

docjfnoel, 2021-07-01:
Thanks again!
I had tried it with no luck.
I'm running Raspbian lite without GUI.
I'm wondering if the problem is that the script starts running before the wifi internet connection is made during the boot up...

llvllch, 2021-07-01:
The script should be clever enough to wait. I'll give it a whirl. Maybe
it's a permissions thing......

bowersbc, 2021-07-02:
My path is /usr/bin/python3. /usr/local/bin/python3 doesn't exist for me.
I tried adding it to cron with this path and it worked:
@reboot /usr/bin/python3 /home/pi/btcticker/btcticker.py

I tried adding it to rc.local and it fails to run and it throws this error:
  File "/home/pi/btcticker/btcticker.py", line 7, in <module>
    import currency
ModuleNotFoundError: No module named 'currency'

Hope this helps

llvllch, 2021-07-02:
but it runs from command line? This looks like an issue with access to the
python modules. My guess is that rc,local is running it as root, and your
modules (specifically nh-currency)  are installed locally. The quick and
dirty fix is to install the module as root with `sudo python3 -m pip
install nh-currency`

docjfnoel, 2021-07-02:
ok after reading your posts and many tries, here's the script that worked for me:

@reboot /usr/bin/python3 //home/pi/btcticker/btcticker.py

llvllch, 2021-07-02:
Thanks for adding your final solution! (Is the double slash on the front of
the path to the script deliberate?)

docjfnoel, 2021-07-02:
The double slash was deliberate but I don't know why I did it...I must have seen the script somewhere 😬
It probably works without it but as they say: if it ain't broke don't fix it 😉

llvllch, 2021-07-03:
Wise words, closing!

closed 2021-07-03 by llvllch