veebch/btcticker #67: Alternative install instructions.

RattyDAVE, 2021-07-04:
These are my install instructions. The ones on your README.me are out of date. As you need to install some packages from a fresh image.

Also you can set the SPI and HOSTNAME from the command line. 

Previous to this I put two files in the /boot ```ssh``` and ```wpa_supplicant.conf``` these are for my headless install.

Once booted ssh into the pi using the username ```pi``` and password ```raspberry``` and run the following.

```
sudo apt-get update
sudo apt-get install -y python3-pip mc git libopenjp2-7
sudo apt-get install -y libatlas-base-dev python3-pil

#set hostname
sudo raspi-config nonint do_hostname pi-ticker

#enable spi (0=on 1=off)
sudo raspi-config nonint do_i2c 0

cd ~
git clone https://github.com/waveshare/e-Paper
git clone https://github.com/llvllch/btcticker.git
cd btcticker
cp -r /home/pi/e-Paper/RaspberryPi_JetsonNano/python/lib .

python3 -m pip install -r requirements.txt

sudo sed -i -e '$i \(while :; do /usr/bin/python3 /home/pi/btcticker/btcticker.py && sleep 120; done) &\n' /etc/rc.local

reboot
```

danijcom, 2021-07-11:
Hi
How did you solve this:
`from waveshare_epd import epd2in7
ModuleNotFoundError: No module named 'waveshare_epd'`

llvllch, 2021-07-11:
Make sure you've done the **Prerequisites** listed at the beginning of the Readme file

RattyDAVE, 2021-07-11:
My Apologises! I copied an old version!!!

cp -r /home/pi/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd .

RattyDAVE, 2021-07-11:
From a fresh image of of the latest version of Raspberry Pi OS Lite from https://www.raspberrypi.org/software/operating-systems/

Previous to this I put two files in the /boot ssh and wpa_supplicant.conf these are for my headless install.

Also you can set the SPI and HOSTNAME from the command line.
Once booted ssh into the pi using the username ```pi``` and password ```raspberry``` and run the following.

```
#!/bin/bash
sudo apt-get update
sudo apt-get install -y python3-pip mc git libopenjp2-7
sudo apt-get install -y libatlas-base-dev python3-pil

#set hostname
sudo raspi-config nonint do_hostname pi-ticker

#enable spi (0=on 1=off)
sudo raspi-config nonint do_i2c 0

cd ~
git clone https://github.com/waveshare/e-Paper
git clone https://github.com/llvllch/btcticker.git
cd btcticker
cp -r /home/pi/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd .

python3 -m pip install -r requirements.txt

sudo sed -i -e '$i \(while :; do /usr/bin/python3 /home/pi/btcticker/btcticker.py && sleep 120; done) &\n' /etc/rc.local

reboot
```

RattyDAVE, 2021-07-11:
> Hi
> How did you solve this:
> `from waveshare_epd import epd2in7 ModuleNotFoundError: No module named 'waveshare_epd'`

I have updated the 1st post with the CORRECT instructions... Sorry!

RattyDAVE, 2021-07-11:
> Make sure you've done the **Prerequisites** listed at the beginning of the Readme file

No need.. Everything is kept in the btcticker directory. INCLUDING  the libraries!

RattyDAVE, 2021-07-11:
> @RattyDAVE This rc.local solution isn't ideal as it is ran as root, better to use **sytemd** with User=pi. The existing instructions seem fine as I followed them just yesterday and ended up with a working ticker.

I aggree the rc.local is not ideal. 

As for the instructions what PiOS image were you using. The LIte version does not have all the packages installed as default.

danijcom, 2021-07-11:
> I aggree the rc.local is not ideal.
>
> As for the instructions what PiOS image were you using. The LIte version does not have all the packages installed as default.

Yes sir. Yesterday I tried to install everything following the instructions in README, but nothing came of it, even though I was still able to install the package waveshare (I installed each library separately, which was required to build the package, it took a long time). I got some errors at startup (something like that .EPD() is not a member of epd2in13_V2). In the end, I gave up and am now trying to blindly follow this instruction. I will definitely edit that message when I get some result.

llvllch, 2021-07-11:
I used PiOS lite, but I installed the extra libraries listed under
*Troubleshooting* in the readme.

You can check for the waveshare library using python3 -m pip list

If it doesnt show:

      cd e-Paper/RaspberryPi_JetsonNano/python
      sudo python3 setup.py install

RattyDAVE, 2021-07-11:
@llvllch These alternative instructions you DONT need to to do the install or install via PIP. Please dont do this with my instructions. As that will cause confusion as the library will be installed in TWO places!

danijcom, 2021-07-11:
The instruction from this post really works. Everything starts up. It's funny, but now I have a different error.
`INFO:root:[Errno 2] No such file or directory` (is happens when script trying to load ethereum logo). I can't see ehtereum.bmp using `ls` in images folder, but 
`>>>os.path.isfile('/home/pi/btcticker/images/currency/ethereum.bmp')
True`
I even tried `sudo rm /home/pi/btcticker/images/currency/ethereum.bmp` it returned no error, the script downloaded the image from coingecko and missed it again :hmm:

RattyDAVE, 2021-07-12:
Nice version with the better startup script.

```
#!/bin/bash
sudo apt-get update
sudo apt-get install -y python3-pip mc git libopenjp2-7
sudo apt-get install -y libatlas-base-dev python3-pil

#set hostname
sudo raspi-config nonint do_hostname pi-ticker

#enable spi (0=on 1=off)
sudo raspi-config nonint do_i2c 0

cd ~
git clone https://github.com/waveshare/e-Paper
git clone https://github.com/llvllch/btcticker.git
cd btcticker
cp -r /home/pi/e-Paper/RaspberryPi_JetsonNano/python/lib/waveshare_epd .
rm -rf /home/pi/e-Paper

python3 -m pip install -r requirements.txt

sudo cat <<EOF > /etc/systemd/system/btcticker.service
[Unit]
Description=btcticker
After=network.target

[Service]
ExecStart=/usr/bin/python3 -u /home/pi/btcticker/btcticker.py
WorkingDirectory=/home/pi/btcticker/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
EOF

sudo systemctl enable btcticker.service
sudo systemctl start btcticker.service

reboot
```

llvllch, 2021-07-12:
Thanks! I'll weave this into the readme. Thanks for sharing!

RattyDAVE, 2021-07-12:
> Thanks! I'll weave this into the readme. Thanks for sharing!

I hope you understand why I installed in a different way. And keeping most installed in /home/pi/btcticker/ 
I find that this makes upgrading easier and also if I want to customise the code everything is in one place. 

I don't want to take anything away of what you have done. But ADD to it. 

Regards!

bowersbc, 2021-07-20:
I had problems with a fresh install following the steps, then followed RattyDAVE's last script manually installing all except for installing as a service. I am able to get it running, but no matter what updatefrequency I choose, it never changes. Any idea on what I can do to get it to cycle? 
Contents:
display:
  cycle: false
  inverted: false
  orientation: 90
  trendingmode: false
  showvolume: false
  showrank: true
ticker:
  currency: ethereum,bitcoin,cardano
  exchange: default
  fiatcurrency: usd
  sparklinedays: 1
  updatefrequency: 10

Here is what I see when it runs:
pi@pi-ticker:~/btcticker $ sudo python3 btcticker.py
INFO:root:epd2in7 BTC Frame
INFO:root:{'display': {'cycle': False, 'inverted': False, 'orientation': 90, 'trendingmode': False, 'showvolume': False, 'showrank': True}, 'ticker': {'currency': 'ethereum,bitcoin,cardano', 'exchange': 'default', 'fiatcurrency': 'usd', 'sparklinedays': 1, 'updatefrequency': 10}}
INFO:root:Throttling update frequency to 180 seconds
INFO:root:Getting Data
INFO:root:https://api.coingecko.com/api/v3/coins/ethereum/market_chart/range?vs_currency=usd&from=1626730690&to=1626817090
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.coingecko.com:443
DEBUG:urllib3.connectionpool:https://api.coingecko.com:443 "GET /api/v3/coins/ethereum/market_chart/range?vs_currency=usd&from=1626730690&to=1626817090 HTTP/1.1" 200 None
INFO:root:Got price for the last 1 days from CoinGecko
INFO:root:https://api.coingecko.com/api/v3/coins/markets?vs_currency=usd&ids=ethereum
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): api.coingecko.com:443
DEBUG:urllib3.connectionpool:https://api.coingecko.com:443 "GET /api/v3/coins/markets?vs_currency=usd&ids=ethereum HTTP/1.1" 200 None
INFO:root:{'id': 'ethereum', 'symbol': 'eth', 'name': 'Ethereum', 'image': 'https://assets.coingecko.com/coins/images/279/large/ethereum.png?1595348880', 'current_price': 1786.96, 'market_cap': 209168700831, 'market_cap_rank': 2, 'fully_diluted_valuation': None, 'total_volume': 20359156767, 'high_24h': 1835.58, 'low_24h': 1728.27, 'price_change_24h': -38.654246154596, 'price_change_percentage_24h': -2.11733, 'market_cap_change_24h': -4663326622.934937, 'market_cap_change_percentage_24h': -2.18084, 'circulating_supply': 116773707.624, 'total_supply': None, 'max_supply': None, 'ath': 4356.99, 'ath_change_percentage': -58.85019, 'ath_date': '2021-05-12T14:41:48.623Z', 'atl': 0.432979, 'atl_change_percentage': 413983.3524, 'atl_date': '2015-10-20T00:00:00.000Z', 'roi': {'times': 79.17163012556685, 'currency': 'btc', 'percentage': 7917.163012556685}, 'last_updated': '2021-07-20T21:37:39.290Z'}
DEBUG:PIL.PngImagePlugin:STREAM b'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM b'tEXt' 41 57
DEBUG:PIL.PngImagePlugin:STREAM b'pHYs' 110 9
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 131 2662
INFO:root:Getting token Image from Image directory
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:Horizontal
DEBUG:root:e-Paper busy
DEBUG:root:e-Paper busy release
DEBUG:root:spi end
DEBUG:root:close 5V, Module enters 0 power consumption ...

llvllch, 2021-07-21:
Nearly there! The config file should say
`cycle: true`

bowersbc, 2021-07-21:
I've tried it both ways, and it will never go to the next coin. I have tried variations of the updatefrequency to no avail.

bowersbc, 2021-07-21:
I changed it to 5 and it took a few minutes, but it finally changed to the next coin.

llvllch, 2021-07-21:
If you set the update frequency to less than 180 (seconds), it sets it to
180 (due to manufacturers recommendations)

llvllch, 2021-07-21:
@RattyDAVE Your install instructions are now in the main Readme... thanks :)

closed 2021-07-21 by llvllch