veebch/btcticker #68: Working with LCD display

chrissteinert, 2021-07-05:
Hi,

I have already attached a 3.5 inch LCD display to my Raspberry and don't want to use the e-INK one. What do I need to change in order to let this code run?

Thanks!
Christian

llvllch, 2021-07-05:
You'll need a way of sending the output image to the display, Depending on
the resolution, you may also need to do some scaling.

RattyDAVE, 2021-07-12:
I have one of these https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A) displays. 
I am using http://www.lcdwiki.com/3.5inch_RPi_Display for the output it is really bad! 

I have almost given up on it. The refresh rate is way too slow to be an enjoyable experience. 

I have not yet tried to integrate it yet. I might have a play!

closed 2021-07-21 by llvllch