veebch/btcticker #70: Shiba Inu shows incorrect dollar value

bowersbc, 2021-07-21:
Coingecko shows Shiba Inu's value at:
$0.00000644
the btcticker shows it at:
$6.44e-06

llvllch, 2021-07-21:
They're the same:
https://www.calculatorsoup.com/calculators/math/scientific-notation-converter.php

closed 2021-07-21 by llvllch

llvllch, 2021-10-21:
It is :) The problem with decimal notation is that you end up writing a lot of zeros, therefore small text and .00000001 does not really look ten times bigger than .000000001. If you really want it, it is easy to change in the code.

PDM78, 2021-11-24:
Sorry for digging up the past but I'd like to know where the code for this is located. Thanks!

veebch, 2021-11-24:
No problem.... it's just in the print statement, but you'll also have a add a conditional font size based on the number of leading zeros. There's a good reason that science likes that notation :)

veebch, 2022-02-05:
The code now scales prices rather than uses scientifc notation.... enjoy :)