veebch/btcticker #71: requirements rpi 404 error

bowersbc, 2021-07-24:
When running this command:
sudo python3 -m pip install -r requirements.txt
it returns the following error:
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://pypi.org/simple/rpi/

When navigating manually to the URL, it does indeed return a 404 Not Found.

llvllch, 2021-08-02:
The use of sudo on the pip install has been deprecated thanks to some improvements by one of the contributors. Try it again with the improved instructions

Kapujino, 2021-08-02:
`pip3 install rpi` still gives a 404, this has nothing to do with sudo.
rpi does not exist (anymore) on https://pypi.org/simple

llvllch, 2021-08-02:
Ah! So it's a pipreqs thing. Fixed! (I think)

Kapujino, 2021-08-02:
Thanks for this repo and the quick fix :+1:

closed 2021-08-04 by llvllch