veebch/btcticker #77: Warning message 'figure.max_open_warning'

thorstenbe, 2021-08-15:
After 15 minutes I've received the following warning message. This might be an issue if it leads to a memory leak but it might also be just a warning. I'm not so into Python therefore I'm just opening the ticket and if it's no issue at all then it can just be closed 😉

```python3[530]: /home/pi/btcticker/btcticker.py:197: RuntimeWarning: More than 20 figures have been opened. Figures created through the pyplot interface (`matplotlib.pyplot.figure`) are retained until explicitly closed and may consume too much memory. (To control this warning, see the rcParam `figure.max_open_warning`).```

llvllch, 2021-08-15:
The closing of images has been a bit of a pain. You may well have came across a bug here. I *think* I fixed it..... try the latest version

thorstenbe, 2021-08-15:
I checked the `git status` and unfortunately it was already the newest version.

llvllch, 2021-08-22:
I used close('all') for the exact same warning on another code yesterday: https://github.com/llvllch/stonks/commit/a59061a0da44c3691c26411a95cfaf62b05bcf7c

commit 5fff86ad (2021-08-22): Fixes #77
Close the fig subplot to prevent memory leak warning message from matplotlib
commit fbff2437 (2021-08-22): Merge pull request #78 from thorstenbe/handle-open-subplot

Fixes #77 Close the fig subplot to prevent memory leak warning message from matplotlib

Tested! Thanks!
closed 2021-08-22 by llvllch