veebch/btcticker #79: How to make it work on a 7.8inch e-Paper HAT

sperez91, 2021-08-24:
Hi all,

I have this 7.8in screen on a RaspberryPi Zero WH: https://www.waveshare.com/7.8inch-e-Paper-HAT.htm

I followed their instructions here: https://www.waveshare.com/wiki/7.8inch_e-Paper_HAT and the tests were successful. Please let me know how difficult it'd be to get it to work on this screen.

Thanks!

llvllch, 2021-08-25:
That's a beauty, Take a look at http://github.com/llvllch/stonks. It should work, and make full use of all that screen.

llvllch, 2021-09-11:
(@txoof is the expert on all things epaper)

txoof, 2021-09-11:
@sperez91 

The way this is written, you'd need to do some maths to get this work on a different display. One option would be to rewrite it to use epdlib. Epdlib allows you to write layouts that are resolution agnostic. That is to say things that use epdlib.Layout() will work on any frame buffered screen at any resolution (within reason). 

[Epdlib](https://github.com/txoof/epdlib/tree/master/epdlib) works with almost everything that waveshare sells currently right out of the box.

If you or @llvllch are interested in porting any of this to Epdlib, I'd be interested in collaborating. I did a quick and dirty proof of concept port of the now defunct [edify/audry](hhttps://github.com/txoof/edify/tree/epdlib_poc) display that will work with almost all of the wavehsare displays including the 7.8 you have. You can get an idea of how easy it is to port to epdlib.

sperez91, 2021-09-25:
I'm a noob ): can't help much. It'd depend on @llvllch 
Anybody else able to help? The Waveshare demo execution works fine on the screen, but I get this error when I try to run the cryptotick.py: "communication with device failed"
[image]

closed 2021-10-16 by llvllch