veebch/btcticker #81: Scale up for a 4.2" waveshare

Zimbakin, 2021-09-11:
Hi, I saw the other link re. 7.2" and the **stonks** app.
However, I prefer the look of your app.

I replaced every instance of `epd2in7` with `epd4in2`
but I just get a thin border around the screen. 4in2 Tests work ok.

Is there any way to simply upscale for a 4.2"? thanks.

closed 2021-09-11 by Zimbakin

PDM78, 2021-11-22:
Just wondering if you found a solution?

Zimbakin, 2021-11-22:
I modified the code as above. Then resized all the elements and made some improvements. If the original dev gives me permission, I’m happy to share it.

Zimbakin, 2021-11-22:
[image]

PDM78, 2021-11-23:
Thank you Zimbakin. What is that .jpeg file? It reroutes back to this thread.

Zimbakin, 2021-11-23:
Tried uploading an image of my 4.2 version. Mobile didn't work for some reason.
[https://drive.google.com/file/d/1h4zB4_ACRvARHb57Q9aXNWMEzUa9tzNo/view?usp=sharing](url)
Hopefully the Dev sees this and grants permission to share.

veebch, 2021-11-23:
Thanks @Zimbakin ....The link is still misbehaving. If you submit a pull request, I can go through the changes.

It seems like a few people are porting to different screen sizes, there must be a relatively elegant way to take screen dimensions as a parameter and have the code take care of all of the scaling.

PDM78, 2021-11-23:
Zimbakin & veebch Thank you.

PDM78, 2021-11-24:
Is there an update to this?