veebch/btcticker #82: Buttons not working

Phozer, 2021-09-12:
Hey

I followed the instructions and got my Ticker to work with no problem. Except for the buttons. No matter when I press them they don't have any effect. Is this a common issue? 

What could I have done wrong? 

Thanks for your help.

reddai75, 2021-09-13:
hello,
I have had  the same problem. After a lot of investigations,
reinstallations etc I concluded that it was a defective display and asked
for a partial refund to the seller.
If you end to a different conclusion please let me know 😉

reddai75, 2021-10-02:
update: I bought a second display, tested on an old raspberry pi b+ and the buttons do not work so there must be something we are missing....

llvllch, 2021-10-02:
I defer to the expertise of @Kapujino, on this one. They looked at the button code in a fair bit of depth in their vast improvement to the computational overhead of the code.

glabbe0299, 2021-10-03:
Same thing here.
buttons do nothing.

llvllch, 2021-10-03:
And the waveshare button demo works?

glabbe0299, 2021-10-03:
Not sure how to run that demo I'm not a programming expert

Kapujino, 2021-10-04:
Created https://github.com/llvllch/btcticker/pull/84 as a workaround. 
You have to pull the repository again to get the latest changes when the PR is merged. (e.g. `cd ~/btcticker` and `git pull`)

@llvllch this workaround causes endless new threads (mentioned in https://github.com/llvllch/btcticker/pull/76#issuecomment-899797910) but the buttons work again.

The "cleaner" solution is to remove the GPIO cleanup from the upstream library.
Edit the file `~/btcticker/waveshare_epd/epdconfig.py` and remove (or comment) the line 87 (`self.GPIO.cleanup()`).
A change of the upstream library was requested here: https://github.com/waveshare/e-Paper/pull/189
Maybe I should create an issue for that.

If this doesn't fix your issue, please do the following and post the output here:

- `sudo systemctl stop btcticker.service` - this stops the probably running service
- `python3 ~/btcticker/btcticker.py --log debug` - this starts the program with verbose logging into your terminal (you can see what is does)

Sorry for the late reply though :)

reddai75, 2021-10-04:
> The "cleaner" solution is to remove the GPIO cleanup from the upstream library. Edit the file `~/btcticker/waveshare_epd/epdconfig.py` and remove (or comment) the line 87 (`self.GPIO.cleanup()`). A change of the upstream library was requested here: [waveshare/e-Paper#189](https://github.com/waveshare/e-Paper/pull/189) Maybe I should create an issue for that.

works great!! thank you!

reddai75, 2021-10-04:
old on... let me investigate... I did a quick check of the buttons, then I left the screen inverted and it seems not updating anymore...

closed 2021-10-04 by llvllch

reddai75, 2021-10-04:
@llvllch I think the problem now is that after pressing the buttons the screen doesn't update anymore

Kapujino, 2021-10-04:
I started the current available version again, cycled a few times through the currencies (always waited until the screen refreshed) and my device is still updating every 300 seconds.

> If this doesn't fix your issue, please do the following and post the output here:
>
>     * `sudo systemctl stop btcticker.service` - this stops the probably running service
>
>     * `python3 [...]

And please keep in mind that the Raspberry Pi Zero and the e-ink display are pretty slow and need some time to update.

reddai75, 2021-10-04:
I have on raspberry pi zero(s) and pi(s) model b+ 1.2, I'll come back to you with logs in the next days if I'm sure it's a software issue, the problem may be on my old b+  ( I need to see what happens on the zeros).
What I'm experiencing with my old b+ is mixed results when I press the buttons...
anyway, thanks for your help!

Phozer, 2021-10-10:
The update fixed the issue for me. Thanks all contributing.