veebch/btcticker #87: FileNotFoundError: [Errno 2] No such file or directory

flyingmonkey12, 2021-10-16:
I feel like I followed every step, but I kept getting these errors...

pi@raspberrypi:~/btcticker $ python3 btcticker.py
sudo: /home/pi/.local/bin/tzupdate: command not found
INFO:root:epd2in7 BTC Frame
INFO:root:{'display': {'cycle': True, 'inverted': False, 'orientation': 90, 'tre                    ndingmode': False, 'showvolume': False, 'showrank': False}, 'ticker': {'currency                    ': 'bitcoin,ethereum,cardano', 'exchange': 'default', 'fiatcurrency': 'usd,eur,g                    bp', 'sparklinedays': 1, 'updatefrequency': 300}}
INFO:root:Getting Data
ERROR:root:[Errno 2] No such file or directory
Traceback (most recent call last):
  File "btcticker.py", line 436, in fullupdate
    display_image(image)
  File "btcticker.py", line 332, in display_image
    epd.Init_4Gray()
  File "/home/pi/btcticker/waveshare_epd/epd2in7.py", line 298, in Init_4Gray
    if (epdconfig.module_init() != 0):
  File "/home/pi/btcticker/waveshare_epd/epdconfig.py", line 76, in module_init
    self.SPI.open(0, 0)
FileNotFoundError: [Errno 2] No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "btcticker.py", line 517, in main
    lastcoinfetch=fullupdate(config,lastcoinfetch)
  File "btcticker.py", line 442, in fullupdate
    display_image(image)
  File "btcticker.py", line 332, in display_image
    epd.Init_4Gray()
  File "/home/pi/btcticker/waveshare_epd/epd2in7.py", line 298, in Init_4Gray
    if (epdconfig.module_init() != 0):
  File "/home/pi/btcticker/waveshare_epd/epdconfig.py", line 76, in module_init
    self.SPI.open(0, 0)
FileNotFoundError: [Errno 2] No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "btcticker.py", line 538, in <module>
    main()
  File "btcticker.py", line 524, in main
    display_image(image)
  File "btcticker.py", line 332, in display_image
    epd.Init_4Gray()
  File "/home/pi/btcticker/waveshare_epd/epd2in7.py", line 298, in Init_4Gray
    if (epdconfig.module_init() != 0):
  File "/home/pi/btcticker/waveshare_epd/epdconfig.py", line 76, in module_init
    self.SPI.open(0, 0)
FileNotFoundError: [Errno 2] No such file or directory

flyingmonkey12, 2021-10-17:
oh boi, now I got this...  I even replaced with a new 2.7 display, wiped out my SD card to start all over again...

pi@raspberrypi:~/btcticker $ python3 -u /home/pi/btcticker/btcticker.py
Matplotlib is building the font cache; this may take a moment.
sudo: /home/pi/.local/bin/tzupdate: command not found
INFO:root:epd2in7 BTC Frame
INFO:root:{'display': {'cycle': True, 'inverted': False, 'orientation': 90, 'trendingmode': False, 'showvolume': False, 'showrank': False, '24h': True}, 'ticker': {'currency': 'bitcoin,ethereum,cardano', 'exchange': 'default', 'fiatcurrency': 'usd,eur,gbp', 'sparklinedays': 1, 'updatefrequency': 300}}
INFO:root:Getting Data
ERROR:root:[Errno 2] No such file or directory
Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 440, in fullupdate
    display_image(image)
  File "/home/pi/btcticker/btcticker.py", line 336, in display_image
    epd.Init_4Gray()
  File "/home/pi/btcticker/waveshare_epd/epd2in7.py", line 298, in Init_4Gray
    if (epdconfig.module_init() != 0):
  File "/home/pi/btcticker/waveshare_epd/epdconfig.py", line 76, in module_init
    self.SPI.open(0, 0)
FileNotFoundError: [Errno 2] No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 521, in main
    lastcoinfetch=fullupdate(config,lastcoinfetch)
  File "/home/pi/btcticker/btcticker.py", line 446, in fullupdate
    display_image(image)
  File "/home/pi/btcticker/btcticker.py", line 336, in display_image
    epd.Init_4Gray()
  File "/home/pi/btcticker/waveshare_epd/epd2in7.py", line 298, in Init_4Gray
    if (epdconfig.module_init() != 0):
  File "/home/pi/btcticker/waveshare_epd/epdconfig.py", line 76, in module_init
    self.SPI.open(0, 0)
FileNotFoundError: [Errno 2] No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/btcticker/btcticker.py", line 542, in <module>
    main()
  File "/home/pi/btcticker/btcticker.py", line 528, in main
    display_image(image)
  File "/home/pi/btcticker/btcticker.py", line 336, in display_image
    epd.Init_4Gray()
  File "/home/pi/btcticker/waveshare_epd/epd2in7.py", line 298, in Init_4Gray
    if (epdconfig.module_init() != 0):
  File "/home/pi/btcticker/waveshare_epd/epdconfig.py", line 76, in module_init
    self.SPI.open(0, 0)
FileNotFoundError: [Errno 2] No such file or directory

hallard, 2021-12-16:
don't want to create another issue but in my case was missing spidev solved by:

`pip3 install spidev`

Worth adding it in [requirements.txt](https://github.com/veebch/btcticker/blob/main/requirements.txt)

closed 2021-12-16 by veebch

hallard, 2021-12-16:
Was a quite old one lite version on my side, not tested on latest lite but won't hurt for sure :-)