veebch/btcticker #92: ‘Issue: Failed to add edge detection’

iiqqrs, 2021-11-14:
At very random times the ticker decides to fail and stare ‘issue: failed to add edge detection Line 521’ 
I simply kill the power and boot it up and it’s back to normal (for some time) and then a few hours… days… it happens again.
[image]

llvllch, 2021-11-15:
I've seen this one but couldn't reproduce it. A quick google made me think it might be a GPIO thing. I'll look into it.

neilneil2000, 2022-01-22:
I forked your project and switched to gpiozero. It seems simpler and more stable than RPi.gpio

veebch, 2022-01-23:
I just glanced at your changes, looks like you have really tidied things up!

neilneil2000, 2022-01-23:
I’ve been teaching myself Python and using it as a little project to hone my skills!

veebch, 2022-01-24:
Great success :)
Closing this issue.... I will look at your code properly and see if I can pick up some good tips!

closed 2022-01-24 by veebch