veebch/btcticker #93: Crypto names..

reddai75, 2021-11-20:
Sometimes the symbol is not enough to remember the name of the crypto. Would it be possible to add the name may be just above the crypto ?
I mean this is a "nice to have" feature:  the code you wrote is already outstanding. 
I wish I could be able to do it by myself... but I have just started learning how to program...

veebch, 2021-11-20:
It's already an option... check the config file :)

reddai75, 2021-11-20:
I checked before asking but I can't find it... may be you forgot to mention it in the config.yaml file ? is not even in the readme file

veebch, 2021-11-20:
You're right. I half remembered that I wrote it into trending mode. It will be easy to add :)

reddai75, 2021-11-20:
wow I managed to do it by myself :)
I added below  
if 'showrank' in config['display'] and config['display']['showrank'] and other['market_cap_rank'] > 1:
            draw.text((10,105),"Rank: " + str("%d" % other['market_cap_rank']),font =font_date,fill = 0)
this:            
draw.text((95,28),whichcoin,font =font_date,fill = 0)

but I'm sure you will have a smarter and nicer solution :))))

veebch, 2021-11-21:
Bravo! 

One comment: In 'trending mode' where it displays 7 trending coins, along with their names, your solution means that the name might display twice. The cleanest way would be to add a new parameter to the config file and then add it in the conditional statement that's already in the code. 

Thinking back, this was not included because I assumed that if a coin was important enough for you to be watching it, that you'd know the symbol. That's why names only (currently) show in trending mode.

closed 2021-11-21 by veebch

reddai75, 2022-04-24:
hehehe it is a bit too much for me :)