veebch/btcticker #94: Issue ‘prices’ Line 436

PDM78, 2021-11-24:
As subject above, I’m getting this error code periodically. 
[image]

Has anyone else encountered this issue?

veebch, 2021-11-25:
The message is a bit cryptic, but it is when there is a problem pulling the data from coingecko. It could be them doing maintenance or blocking you for too many requests. I noticed them doing a temporary block on my IP if I suddenly started hitting their api fast (seconds apart, multiple script runs).

PDM78, 2021-11-25:
Got it! Been testing since yesterday and seldom does the message appear. I have the display rotating every 30 seconds.

reddai75, 2021-11-25:
it's normal if you update every 30 sec. Don't forget that coingecko API is
a free service as long as you don't abuse it.

veebch, 2021-11-25:
a consideration is that the lifespan of epaper is measured in refreshes, so an epaper refreshing every two minutes will last 4 times longer than one refreshing every 30 seconds

reddai75, 2021-11-25:
> a consideration is that the lifespan of epaper is measured in refreshes, so an epaper refreshing every two minutes will last 4 times longer than one refreshing every 30 seconds

about that: it would be nice to have the possibility to add a timeframe for sleepmode, like during the night!

sterdeus, 2021-11-25:
For consideration of energy savings, I think a sleep mode would be negligible. As mentioned, lifespan of the screen would be the main purpose. However, I find myself waking up in the middle of the night and quickly checking bitcoin's price. FOMO to the max.

veebch, 2021-11-27:
This is the main use-case for me. I dream about a price, wake up, glance at the ticker, see the real price

closed 2021-11-27 by veebch