veebch/btcticker #96: Currency Symbol position

hallard, 2021-12-16:
Not an issue but an option to set the currency symbol position would be great (I dirty hacked code, not clean).

I mean in EU we use `12345.12€` notation, not `€12345.12` but if we display USD same `12345.12$` not `$12345.12` 

Letting user to select as an option to put symbol before or after would be perfect

veebch, 2022-02-05:
This could be set as a configuration option. That said, I've never seen the Euro symbol as a suffix. I know it is really common for things like the Czech Crown... I will look at the currency module and see whether it contains such info.

veebch, 2022-02-05:
Done.... to switch to an alternative locale, you can edit line 283. I've tested it on a few coins and it looks good. Closing for now, let me know if you have any issues with it :)

closed 2022-02-05 by veebch