veebch/btcticker #101: Font size problem

SzilagyiR, 2022-03-28:
Hi!

Thanks for your wok, it's amazing!

I'm trying to show the price in HUF, but it contains too many number: HUF16,286975
How can I reduce the font size, to fit it on the screen?

Thanks!

veebch, 2022-03-28:
It's good to see that prices are getting too big for the screen :)
I will try to make it automatic as soon as I get a chance. For a quick fix, line 315. Change from 50 to a smaller number :)

SzilagyiR, 2022-03-28:
Thanks for the fast solution, it is perfect now! Font size is 35

[image]

closed 2022-03-28 by SzilagyiR

veebch, 2022-03-28:
Excellent.... another line that might interest you is 283, where you can change the locale.

SzilagyiR, 2022-03-28:
More than perfect! :)
[image]

veebch, 2022-03-28:
Excellent. And for full satisfaction, pull gently on the green tab to remove the screen protector :)

SzilagyiR, 2022-03-28:
Sure, but first I print it a nice house for it with a screen protector :)

veebch, 2022-03-28:
I just thought I'd mention it because I didn't realise it was there for ages :) Enjoy watching the price!

veebch, 2022-03-29:
I've just added locale to the config file and an automatic scaling. Not tested yet, that's after coffee

SzilagyiR, 2022-03-29:
Don't feel like implement this project into yours?
https://github.com/aCallum/NicePi

It would be good because don’t need a separate PI for it. :)

veebch, 2022-03-29:
Funnily enough, this was something I was looking at doing.

The locale addition to the config file is a good one. Thanks for reminding me to do it.