veebch/btcticker #102: Divide by zero bug

adamneilson, 2022-04-14:
The last few days I've been experiencing an issue with my beloved ticker where it crashes with an error "Issue float division by zero Line: 450"

I'm tracking the following: BTC, ALGO and XAU

I have to restart the pi to get it running again. I'm not sure what other details I can give as there's not much to display on the epaper screen but happy to elaborate if needed.

veebch, 2022-04-14:
I've seen that error on old code. Have you pulled the latest version. That should at least give a better error message if it does fail

adamneilson, 2022-04-14:
i just pulled latest updating from 13566ee to f339809. Fingers crossed!

adamneilson, 2022-04-14:
Two hours later and no further occurrence so I'll close this issue.

adamneilson, 2022-04-15:
Unfortunately the problem still exists. The error changed slightly but still not very verbose:

_"Issue: float division by zero Line: 454"_

veebch, 2022-04-15:
Useful! It's happening in MakeSpark. I will look into it

veebch, 2022-04-15:
I have a copy running with your configuration on. I will try to reproduce the error

adamneilson, 2022-04-15:
👍

adamneilson, 2022-04-15:
Oh! I just checked my conf and it's actually PAXG not XAU. So my config is:
```
currency: algorand,bitcoin,pax-gold
```

veebch, 2022-04-15:
OK..... I will change it. Most likely it was this:

https://status.coingecko.com/incidents/81704

veebch, 2022-04-17:
Still running here. What about you?

adamneilson, 2022-04-17:
Yeah, it's running well here. I guess it must have been coingecko API problems. It might be good to catch the API connection error and display an appropriate message on screen but that's definitely a nice to have rather than an essential :)

veebch, 2022-04-17:
Yeah, the error trapping could definitely be tidied. It's on the to-do list
:)

closed 2022-04-17 by veebch

adamneilson, 2022-04-19:
https://status.coingecko.com is saying there's no issues but I've been having the same error as before this afternoon. @veebch is yours working okay?

veebch, 2022-04-19:
It is.... What are your coins and refresh time?

adamneilson, 2022-04-20:
currency: algorand,bitcoin,pax-gold with refresh time of 120 seconds

veebch, 2022-04-20:
I've not seen any problems yet. I'll keep an eye on it. Regardless, the error trapping sucks..... moving it up my to-do list

adamneilson, 2022-04-22:
Is there a log file somewhere that I could grep?

veebch, 2022-04-22:
You could check dmesg or just run it interactively from the terminal with the debug flag:
`python3 btcticker.py --log DEBUG`